kizniche/Mycodo #865: Trigger: timer does not work with PWP-output

grux77, 2020-10-11:
Linux RPi4 5.4.51-v7l+
Mycodo Version: 8.8.6
Python Version: 3.7.3 (default, Jul 25 2020, 13:03:44) [GCC 8.3.0]
Database Version: 0e150fb8020b
Daemon RAM Usage: 53.58 MB
Frontend RAM Usage: 54.272 MB

I upgraded mycodo from 8.5.8 to 8.8.6. I use the Trigger: timer to switch the LED lighting in the aquarium. Everything works like before, only the triggers don't want to switch the PWM output.
No matter if daily span or duration. When I switch the PWM output manually, the LEDs light up.
The trigger switches ON/Off GPIO-output without any problems. It looks like the trigger doesn't like PWM outputs.
Debug log does not show any errors. Hardware is ok. I have the error on two pi boards.
I have already tried several PWM-hardware-pins and also Software-PWM-Pins: 
manually everything is ok but the trigger timer does not switch.
Does anyone else have the problem or do I have to make additional settings in 8.8.6? 
I've already reinstalled everything - without success.

kizniche, 2020-10-12:
Thanks for the bug report. I'll look into it.

kizniche, 2020-10-19:
I just pushed a fix that will be available in the next release.

grux77, 2020-10-25:
Hello Kyle!
I tested your master version this week. The Trigger:Timer works again with PWM-Output.
Good job as always!
Now I want to switch my LCD 16x2 on and off with a button. I have created an output 'On / Off GPIO (RPi.GPIO)' on PIN 23.
The pin goes HIGH when I press the button. Then I created the "Trigger: Output (ON/OFF)": "If State" is "ON (no duration)" or "ON (any duration)" 
The trigger turns on the backlight of the LCD when the output is active.
The problem: when I physically press the button, I see in the WebUI that the output is active, but the trigger does nothing. 
If I activate the output in the WebUI then the trigger reacts. Strange! 
Could you check it out when you have time?

Here is the log when I activate the output on the web. Debug is also activated in the trigger, but it shows nothing even though it is triggered:
2020-10-25 18:21:34,290 - DEBUG - mycodo.controllers.controller_output - output_on_off(bc032a59-3472-4d8b-ae5c-d690285a20d6, on, 0, sec, 0.0, 0.0, True)
2020-10-25 18:21:34,291 - DEBUG - mycodo.outputs.on_off_gpio_bc032a59 - output_on_off(on, 0, sec, 0.0, 0.0, True)
2020-10-25 18:21:34,291 - DEBUG - mycodo.outputs.on_off_gpio_bc032a59 - Output bc032a59-3472-4d8b-ae5c-d690285a20d6 CH0 (Button-ON) ON at 2020-10-25 18:21:34.291306. Output returned: success
2020-10-25 18:21:35,832 - DEBUG - mycodo.controllers.controller_output - output_on_off(bc032a59-3472-4d8b-ae5c-d690285a20d6, off, 0, None, 0.0, 0.0, True)
2020-10-25 18:21:35,833 - DEBUG - mycodo.outputs.on_off_gpio_bc032a59 - output_on_off(off, 0, None, 0.0, 0.0, True)
2020-10-25 18:21:35,833 - DEBUG - mycodo.outputs.on_off_gpio_bc032a59 - Output bc032a59-3472-4d8b-ae5c-d690285a20d6 CH0 (Button-ON) OFF at 2020-10-25 18:21:35. Output returned: success

When I physically press the button, nothing is displayed in the log, although "Log Level: Debug" is activated in the output. And I can see on the WebUI that it is active.

grux77, 2020-10-25:
I've upgraded to master. It works fine again. Very good job as always!

kizniche, 2020-10-27:
Please keep discussions of issues to their respective thread: #842

closed 2020-10-27 by kizniche