kizniche/Mycodo #866: Modifying an input results in: Error: Modify Input: Expecting value: line 1 column 1 (char 0)

kpslc, 2020-10-17:
### Attempting to modify a few of the inputs that were active results in the error:

Error: Modify Input: Expecting value: line 1 column 1 (char 0)

Tried with the Atlas PT-1000 and with a GPIO state monitor (for our water valve). Tried to change the period to a different value. Clicked "save" and the error displays.

Also tried changing the name of the device as a test without changing any other value. Same error.

It is strange that the pH and EC data input items allowed either value (name or period) to be changed.

### Versions:

 - Mycodo Version: 8.8.6
 - Raspberry Pi Version: 4
 - Mycodo Version: 8.8.6 
 -Python Version: 3.7.3 (default, Jul 25 2020, 13:03:44) [GCC 8.3.0] 
 -Database Version: 0e150fb8020b
 -Daemon Status: Running 
 -Frontend Process ID: 485 
 -Frontend RAM Usage: 79.72 MB 
 -Frontend Virtualenv: Yes
Kernel Information: Linux Aeroponics 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux

kizniche, 2020-10-19:
I am unable to reproduce this issue:

1. Select GPIO State input and click Add.
2. Expand new GPIO State input and click Save. (saves fine)
3. Expand new GPIO State input, change period and name, and click Save. (saves fine)

kizniche, 2020-10-20:
Likely a duplicate of #868 but this issue I couldn't replicate.

kpslc, 2020-10-20:
I tried the one step suggested in #866 of adding a convert measurement, then tried to modify the GPIO State object. It didn't help. Link to [movie](https://drive.google.com/file/d/1WCUxekHHhrxWEx2hG3iURBmPEjqlWMpG/view?usp=sharing) showing error. 
Wondering what your code is looking for when the save button is clicked where it is expecting a value that isn't there?
I have three systems setup that are all doing the same thing. They were all built from scratch, not restored from a backup.

kizniche, 2020-10-20:
I think I figured out the issue. It appears to only affect Inputs that were created a relatively long time ago. I suspect you did not test creating a new Input (which should not display this issue). Regardless, I just pushed a fix that should resolve this issue for your older Inputs, that wiil be available in the next release. If you want to test before the release, you will have to [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch).

kpslc, 2020-10-20:
Thank you! I did create a new one and as you guessed, it didn’t have the problem. Thanks for the help.
I’ll wait for the next release to update these systems.

Kevin Plumb
C: 801.414.8643
E: kevin.plumb@gmail.com <mailto:kevin.plumb@gmail.com>

closed 2020-10-27 by kizniche