kizniche/Mycodo #867: compatible with DFRobot sensor/controller or other Chinese made sensors

chamroeunbm, 2020-10-18:
**Is your feature request related to a problem? Please describe.**
Altas scientific is too costly even though it is very accurate and stable so i prefer cheaper one like DFRobot.

**Describe the solution you'd like**
will mycodo work with product from DFRobot or other Chinese brand? i am not sure if they are all standardised output.

kizniche, 2020-10-19:
Possibly. Without mentioning a specific product, that's the best answer I can give.

chamroeunbm, 2020-10-20:
Oh sorry,
Here's the list of the probe:

https://www.dfrobot.com/product-2069.html
https://www.dfrobot.com/product-1123.html

kizniche, 2020-10-20:
Those sensors output a voltage from 0 to 3 and 3.4 volts, so since Raspberry PIs can't measure voltages, you'll need an analog-to-digital converter capable of accepting those output ranges. Since most ADCs have multiple channels, you could use the same ADC to connect both sensors. See the ADC section of the manual for supported ADCs: https://kizniche.github.io/Mycodo/Analog-To-Digital-Converters/

Additionally, you can always build and import your own custom Input Module if you have another ADC you want to use with Mycodo.

chamroeunbm, 2020-10-22:
Sorry, my background is mobile app/web app development(C,C++, Objective-C, Java) thus I have no or little knowledge on this thing.

Here they have their Raspberry code in Python: https://github.com/DFRobot/DFRobot_PH/tree/master/RaspberryPi/Python
DO you think this will work with your code or require a lot of effort?

kizniche, 2020-11-07:
That code is merely reading as ADS1115 ADC ("The Analog pH Sensor should work with ADS1115"). As I mentioned previously, any ADC can measure the voltage output of the sensor you linked. If you look at the manual link I provided, you'll see the ADS1115 is supported by Mycodo.

chamroeunbm, 2020-11-12:
I got my first custom input EC/PH working now even though my code look ugly due to my zero experience in Python.

Many thanks for your support and this great project.

closed 2020-11-12 by chamroeunbm

kizniche, 2020-11-12:
Would you be willing to share your code, so it could possibly be included in the built-in set of inputs? Don't worry about what it looks like, we've all started somewhere,and it's quite impressive to have made a custom input having no Python experience.

kizniche, 2020-11-12:
Also, if you have any feedback about the documentation, I'd like to hear what you thought could possibly be improved or made clearer.

chamroeunbm, 2020-11-17:
I updated your existing atlas_EC input code and use their sample python code. Now I am working on getting the the low cost peristaltic pump to work and will get back to  write code for calibration.

# coding=utf-8
import copy
import time
import sys
from flask_babel import lazy_gettext

from mycodo.inputs.base_input import AbstractInput
from mycodo.inputs.sensorutils import calculate_dewpoint
from mycodo.utils.DFRobot_ADS1115 import ADS1115
from mycodo.utils.DFRobot_EC      import DFRobot_EC
from mycodo.utils.DFRobot_PH      import DFRobot_PH
from mycodo.utils.system_pi import str_is_float

ADS1115_REG_CONFIG_PGA_6_144V        = 0x00 # 6.144V range = Gain 2/3
ADS1115_REG_CONFIG_PGA_4_096V        = 0x02 # 4.096V range = Gain 1
ADS1115_REG_CONFIG_PGA_2_048V        = 0x04 # 2.048V range = Gain 2 (default)
ADS1115_REG_CONFIG_PGA_1_024V        = 0x06 # 1.024V range = Gain 4
ADS1115_REG_CONFIG_PGA_0_512V        = 0x08 # 0.512V range = Gain 8
ADS1115_REG_CONFIG_PGA_0_256V        = 0x0A # 0.256V range = Gain 16

def constraints_pass_positive_value(mod_input, value):
    """
    Check if the user input is acceptable
    :param mod_input: SQL object with user-saved Input options
    :param value: float or int
    :return: tuple: (bool, list of strings)
    """
    errors = []
    all_passed = True
    # Ensure value is positive
    if value <= 0:
        all_passed = False
        errors.append("Must be a positive value")
    return all_passed, errors, mod_input

# Measurements
measurements_dict = {
    0: {
        'measurement': 'electrical_conductivity',
        'unit': 'uS_cm'
    }
}
# Input information
# See the inputs directory for examples of working modules.
# The following link provides the full list of options with descriptions:
# https://github.com/kizniche/Mycodo/blob/single_file_input_modules/mycodo/inputs/examples/example_all_options_temperature.py
INPUT_INFORMATION = {
    'input_name_unique': 'dfrobot_ec_03',
    'input_manufacturer': 'AAA Company X',
    'input_name': 'DFRobot EC',
    'input_library': 'pylibftdi/fcntl/io/serial',
    'measurements_name': 'Electrical Conductivity',
    'measurements_dict': measurements_dict,
    'measurements_use_same_timestamp': True,

    'url_manufacturer': 'https://www.st.com/en/imaging-and-photonics-solutions/vl53l0x.html',
    'url_datasheet': 'https://www.st.com/resource/en/datasheet/vl53l0x.pdf',
    'url_product_purchase': [
        'https://www.adafruit.com/product/3317',
        'https://www.pololu.com/product/2490'
    ],

   'options_enabled': [
        'i2c_location',
        'period',
        'pre_output'
    ],
    'options_disabled': ['interface'],
[... 76 lines cut ...]

    def get_measurement(self):
        """ Measures temperature and humidity """
        electrical_conductivity = None
        self.return_dict = copy.deepcopy(measurements_dict)

        # Compensate measurement based on a temperature measurement
        #if self.temperature_comp_meas_measurement_id:
        self.logger.debug("EC Sensor set to get data")
        #Read your temperature sensor to execute temperature compensation
        temperature = 25
    #Set the IIC address
        adc0 = self.ads1115.readVoltage(0)
        
    #Convert voltage to EC with temperature compensation
        EC = self.ec_device.readEC(adc0['r'],temperature)
        #PH = ph.readPH(adc1['r'],temperature)
        self.logger.debug("Temperature:%.1f ^C EC:%.2f ms/cm " %(temperature,EC))
        electrical_conductivity = float(EC)
        self.value_set(0, electrical_conductivity)

        return self.return_dict
# 
#     def button_one(self, args_dict):
#         self.logger.error("Button One Pressed!: {}".format(int(args_dict['button_one_value'])))