kizniche/Mycodo #868: Atlas ORP I2C: problem adding new input; once added the input returns the linked temp compensation value

gtnorman, 2020-10-18:
Please DO NOT OPEN AN ISSUE:

 - If your Mycodo version is not the latest release version, please update your device before submitting your issue (unless your issue is related to not being able to upgrade). Your problem might already be solved.
 - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue.

Please complete as many of the sections below, if applicable, to provide the most information that may help with investigating your issue. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help those providing assistance by reducing the time required to help you.

### Describe the problem/bug

A clear and concise description of what you were trying to do and what the problem/bug is.

### Versions:

 - Mycodo Version: [e.g. 8.0.3]
 - Raspberry Pi Version: [e.g. 3B+]
 - Raspbian OS Version: [e.g. Buster Lite]

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Expected behavior

A clear and concise description of what you expected to happen.

### Screenshots

If applicable, add screenshots to help explain your problem.

### Additional context

Is there anything that should be added to make it easier to address this issue?

gtnorman, 2020-10-18:
When adding Atlas ORP I2C input, the setup is saved successfully but strips out the temperature compensation source and max age unless a Convert Measurements Name is also entered.  This fault is reproducible.

1. Add I2C Atlas ORP
2. Complete setup using Atlas PT-1000 as temp comp source, 120 seconds max age.
3. Save.
4. Click Activate, receive error message _Temperature Compensation Max Age is required to be set_
5. Open setup, see that the temp comp and max age are missing.  Enter values and save successfully.
6. Click Activate, receive error message _Temperature Compensation Max Age is required to be set_

The only way to successfully save the Atlas ORP setup is to add a Convert Measurements Name value.  Then the temp comp selection and max age will be saved. 

[image]
[image]
[image]
[image]
[image]
[image]

gtnorman, 2020-10-18:
Sorry Kyle, left out this is 8.8.6 and up to date Raspbian.  The ORP sensor is of a type I've successfully used as a swimming pool sensor.  It is not purchased through Atlas.  The EZO board has been placed in I2C mode via hardware reset and is blue.  The EZO resides on a Whitebox Labs Tentacle [Tentacle](https://atlas-scientific.com/carrier-boards/whitebox-labs-tentacle-t3/#:~:text=Whitebox%20Labs%20Tentacle%20T3%20%20%204%2B%20units,%20%2025%2B%20units%20%20%20%2494.50%20) on one of the two isolated circuits.  This same Tentacle circuit works successfully with an EZO pH board.  

Once the Atlas ORP setup is saved by adding a Convert Measurement _Name_, the activated ORP input does not appear to be successfully querying the EZO ORP board; the LED remains solid blue and never flashes green to indicate a measurement is being made.  The Live data reveals that the Atlas PT-1000 temperature compensation measurement is being reported as the ORP value:
[image]

kizniche, 2020-10-18:
Strange. Thanks for the detailed info. I'll check it out.

gtnorman, 2020-10-19:
Kyle, there's nothing wrong with the ORP processing.  I'm sorry I didn't make the basic check of reviewing the I2C addresses.

It seems weirdly intermittent, but the ORP I2C address is not being saved after being changed during setup.  The default address is 0x66 when the ORP probe is first added and has to be changed to 0x62 which is Atlas's default ORP probe address.  There is a path where this value isn't updated (start with 0x66, change to 0x62, Save, open and see that address is still 0x66).

I'll play with the workflow this evening until I can reliably reproduce the issue.

So...everything appears fine with the actual processing; it's just a setup issue.

kizniche, 2020-10-19:
Thanks for the update. That makes a lot more sense. I should be able to investigate myself today or tomorrow.

kizniche, 2020-10-19:
I found the issue, and it stemmed from the Temperature Compensation Max Age not receiving the default value upon the Input's creation. I just pushed a fix that will be available in the next release. In the meantime, if you just make sure that input field has a numerical value, saving should work.

gtnorman, 2020-10-20:
Thanks for the instant fix!  Change the default address to 0x62 while you’re at it 👍

gtnorman, 2020-10-21:
Thanks Kyle

closed 2020-10-27 by kizniche