kizniche/Mycodo #871: OLED with new circuit python

gtnorman, 2020-10-29:
### Describe the problem/bug

OLED display is not working when configured with circuit python drivers

### Versions:
Raspberry pi 4B 8Gb
Mycodo Version: 8.8.7
Python Version: 3.7.3 (default, Jul 25 2020, 13:03:44) [GCC 8.3.0]
Database Version: 66e27f22b15a
Daemon Status: Running
Frontend Process ID: 522
Frontend RAM Usage: 59.012 MB
Frontend Virtualenv: Yes
uptime
 11:42:40 up 15 min,  2 users,  load average: 1.06, 1.23, 0.95
Kernel Information: uname -a
Linux raspberrypi 5.4.72-v7l+ #1356 SMP Thu Oct 22 13:57:51 BST 2020 armv7l GNU/Linux

### Reproducibility

1. Deleted old OLED output setup which used the deprecated drivers.  This output has been deactivated since failed attempts to get this feature to work.
2. Added new 128x64 OLED output using the new circuit python drivers.  
3. Configured and activated the output.  Confirmed I2C address.
4. No display
5. Reboot
6. No display
7. Deactivate 128x64 output
8. Add 128x32 OLED output using the new circuit python drivers. 
9. Configured and activated the output
10. No display
11. Reboot
12. No display

### Expected behavior

Configured data lines appear on OLED display

### Screenshots

[image]
[image]

### Additional context

2020-10-29 11:29:22,157 - ERROR - mycodo.controllers.controller_lcd_d233f2d9 - initialize_variables() Exception: 'SSD1306_I2C' object has no attribute 'begin'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 72, in run
    self.initialize_variables()
  File "/var/mycodo-root/mycodo/controllers/controller_lcd.py", line 297, in initialize_variables
    self.lcd_out = LCD_Pioled_Circuitpython(lcd_dev)
  File "/var/mycodo-root/mycodo/devices/lcd_pioled_circuitpython.py", line 60, in __init__
    self.disp.begin()
AttributeError: 'SSD1306_I2C' object has no attribute 'begin'
2020-10-29 11:29:22,158 - INFO - mycodo.controllers.controller_lcd_d233f2d9 - Activated in 971.4 ms
2020-10-29 11:30:12,211 - ERROR - mycodo.controllers.controller_lcd_ca044c92 - initialize_variables() Exception: 'SSD1306_I2C' object has no attribute 'begin'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 72, in run
    self.initialize_variables()
  File "/var/mycodo-root/mycodo/controllers/controller_lcd.py", line 297, in initialize_variables
    self.lcd_out = LCD_Pioled_Circuitpython(lcd_dev)
  File "/var/mycodo-root/mycodo/devices/lcd_pioled_circuitpython.py", line 60, in __init__
    self.disp.begin()
AttributeError: 'SSD1306_I2C' object has no attribute 'begin'
2020-10-29 11:30:12,212 - INFO - mycodo.controllers.controller_lcd_ca044c92 - Activated in 507.5 ms

kizniche, 2020-10-29:
Duplicate of #842

closed 2020-10-29 by kizniche

gtnorman, 2020-10-30:
Sorry Dr G.  I'll be more careful in the future.  Don’t wash away!

Greg Norman
RM Systems
+1 (847) 816-0040 office
+1 (847) 804-7509 cell

Closed #871<https://github.com/kizniche/Mycodo/issues/871>.