kizniche/Mycodo #874: front end crash after removing custom input

rbultman, 2020-11-05:
The crash was immediate.  I clicked "delete" on the custom sensor, and the below traceback appeared.

Here is the traceback:
Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 639, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 1173, in page_live
    list_devices_adc=list_analog_to_digital_converters(),
  File "/home/pi/Mycodo/mycodo/utils/inputs.py", line 61, in list_analog_to_digital_converters
    dict_inputs = parse_input_information()
  File "/home/pi/Mycodo/mycodo/utils/inputs.py", line 109, in parse_input_information
    input_custom = load_module_from_file(full_path, 'inputs')
  File "/home/pi/Mycodo/mycodo/utils/modules.py", line 16, in load_module_from_file
    module_custom = importlib.util.module_from_spec(spec)
  File "<frozen importlib._bootstrap>", line 580, in module_from_spec
AttributeError: 'NoneType' object has no attribute 'loader'

rbultman, 2020-11-05:
I ran the frontend in debug bug mode and immediately got the following error in the terminal:

Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/start_flask_ui.py", line 11, in <module>
    from mycodo.mycodo_flask.app import create_app
  File "/var/mycodo-root/mycodo/mycodo_flask/app.py", line 32, in <module>
    from mycodo.mycodo_flask import routes_page
  File "/var/mycodo-root/mycodo/mycodo_flask/routes_page.py", line 90, in <module>
    from mycodo.mycodo_flask.forms import forms_input
  File "/var/mycodo-root/mycodo/mycodo_flask/forms/forms_input.py", line 34, in <module>
    class InputAdd(FlaskForm):
  File "/var/mycodo-root/mycodo/mycodo_flask/forms/forms_input.py", line 36, in InputAdd
    dict_inputs = parse_input_information()
  File "/var/mycodo-root/mycodo/utils/inputs.py", line 109, in parse_input_information
    input_custom = load_module_from_file(full_path, 'inputs')
  File "/var/mycodo-root/mycodo/utils/modules.py", line 16, in load_module_from_file
    module_custom = importlib.util.module_from_spec(spec)
  File "<frozen importlib._bootstrap>", line 580, in module_from_spec
AttributeError: 'NoneType' object has no attribute 'loader'

kizniche, 2020-11-05:
Can you provide note information? Mycodo version, input type, etc.

rbultman, 2020-11-05:
The Mycodo version is 8.8.8.  It's a new setup on a Pi 3.  The input is a custom input I'm working on to integrate sensors via the Grove Pi+ hat.  It's an input I'm trying to test and debug.  I posted on the forum for help leaning how to test custom inputs.

kizniche, 2020-11-05:
Can you attach the custom module you used that this error happened with so I can test? At the very least, I can catch those errors so it doesn't cause the frontend crash, but I can also look at your module for any issues I see that may be the cause.

rbultman, 2020-11-05:
Here it is:
[grove_temp_humidity.zip](https://github.com/kizniche/Mycodo/files/5495859/grove_temp_humidity.zip)

rbultman, 2020-11-05:
This is the dependency.
[grovepi.zip](https://github.com/kizniche/Mycodo/files/5495940/grovepi.zip)

rbultman, 2020-11-05:
More details:
uname:  Linux raspberrypi 5.4.72-v7+ #1356 SMP Thu Oct 22 13:56:54 BST 2020 armv7l GNU/Linux
OS: raspios buster 2020-08-20 full

I made a fresh image on an SD card, then installed the [Grove Pi+ firmware](https://www.dexterindustries.com/GrovePi/get-started-with-the-grovepi/) following the advanced instructions.  I verified that I could read from the sensor using the example code from the GrovePi installation.  Then I installed the Mycodo firmware per your instructions.

After the failure, I wiped the Mycodo directory and reinstalled it.  Without adding any sensor or making any changes, the front end has crashed again.  Here is the output:
pi@raspberrypi:~ $ journalctl -u mycodoflask | tail -n 50
Nov 05 14:41:12 raspberrypi gunicorn[453]: [2020-11-05 14:41:12 -0500] [453] [INFO] Listening at: unix:/var/run/mycodoflask.sock (453)
Nov 05 14:41:12 raspberrypi gunicorn[453]: [2020-11-05 14:41:12 -0500] [453] [INFO] Using worker: gthread
Nov 05 14:41:12 raspberrypi gunicorn[453]: [2020-11-05 14:41:12 -0500] [623] [INFO] Booting worker with pid: 623
Nov 05 14:41:19 raspberrypi gunicorn[453]: [2020-11-05 14:41:19 -0500] [623] [ERROR] Exception in worker process
Nov 05 14:41:19 raspberrypi gunicorn[453]: Traceback (most recent call last):
Nov 05 14:41:19 raspberrypi gunicorn[453]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/arbiter.py", line 583, in spawn_worker
Nov 05 14:41:19 raspberrypi gunicorn[453]:     worker.init_process()
Nov 05 14:41:19 raspberrypi gunicorn[453]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/workers/gthread.py", line 92, in init_process
Nov 05 14:41:19 raspberrypi gunicorn[453]:     super().init_process()
Nov 05 14:41:19 raspberrypi gunicorn[453]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/workers/base.py", line 119, in init_process
Nov 05 14:41:19 raspberrypi gunicorn[453]:     self.load_wsgi()
Nov 05 14:41:19 raspberrypi gunicorn[453]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/workers/base.py", line 144, in load_wsgi
Nov 05 14:41:19 raspberrypi gunicorn[453]:     self.wsgi = self.app.wsgi()
Nov 05 14:41:19 raspberrypi gunicorn[453]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/app/base.py", line 67, in wsgi
Nov 05 14:41:19 raspberrypi gunicorn[453]:     self.callable = self.load()
Nov 05 14:41:19 raspberrypi gunicorn[453]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/app/wsgiapp.py", line 49, in load
Nov 05 14:41:19 raspberrypi gunicorn[453]:     return self.load_wsgiapp()
Nov 05 14:41:19 raspberrypi gunicorn[453]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/app/wsgiapp.py", line 39, in load_wsgiapp
Nov 05 14:41:19 raspberrypi gunicorn[453]:     return util.import_app(self.app_uri)
Nov 05 14:41:19 raspberrypi gunicorn[453]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/util.py", line 358, in import_app
Nov 05 14:41:19 raspberrypi gunicorn[453]:     mod = importlib.import_module(module)
Nov 05 14:41:19 raspberrypi gunicorn[453]:   File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
Nov 05 14:41:19 raspberrypi gunicorn[453]:     return _bootstrap._gcd_import(name[level:], package, level)
Nov 05 14:41:19 raspberrypi gunicorn[453]:   File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
Nov 05 14:41:19 raspberrypi gunicorn[453]:   File "<frozen importlib._bootstrap>", line 983, in _find_and_load
Nov 05 14:41:19 raspberrypi gunicorn[453]:   File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
Nov 05 14:41:19 raspberrypi gunicorn[453]:   File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
Nov 05 14:41:19 raspberrypi gunicorn[453]:   File "<frozen importlib._bootstrap_external>", line 728, in exec_module
Nov 05 14:41:19 raspberrypi gunicorn[453]:   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
Nov 05 14:41:19 raspberrypi gunicorn[453]:   File "/home/pi/Mycodo/mycodo/start_flask_ui.py", line 11, in <module>
Nov 05 14:41:19 raspberrypi gunicorn[453]:     from mycodo.mycodo_flask.app import create_app
Nov 05 14:41:19 raspberrypi gunicorn[453]:   File "/home/pi/Mycodo/mycodo/mycodo_flask/app.py", line 32, in <module>
Nov 05 14:41:19 raspberrypi gunicorn[453]:     from mycodo.mycodo_flask import routes_page
Nov 05 14:41:19 raspberrypi gunicorn[453]:   File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 90, in <module>
Nov 05 14:41:19 raspberrypi gunicorn[453]:     from mycodo.mycodo_flask.forms import forms_input
Nov 05 14:41:19 raspberrypi gunicorn[453]:   File "/home/pi/Mycodo/mycodo/mycodo_flask/forms/forms_input.py", line 34, in <module>
Nov 05 14:41:19 raspberrypi gunicorn[453]:     class InputAdd(FlaskForm):
Nov 05 14:41:19 raspberrypi gunicorn[453]:   File "/home/pi/Mycodo/mycodo/mycodo_flask/forms/forms_input.py", line 36, in InputAdd
Nov 05 14:41:19 raspberrypi gunicorn[453]:     dict_inputs = parse_input_information()
Nov 05 14:41:19 raspberrypi gunicorn[453]:   File "/home/pi/Mycodo/mycodo/utils/inputs.py", line 109, in parse_input_information
Nov 05 14:41:19 raspberrypi gunicorn[453]:     input_custom = load_module_from_file(full_path, 'inputs')
Nov 05 14:41:19 raspberrypi gunicorn[453]:   File "/home/pi/Mycodo/mycodo/utils/modules.py", line 16, in load_module_from_file
Nov 05 14:41:19 raspberrypi gunicorn[453]:     module_custom = importlib.util.module_from_spec(spec)
[... cut ...]
Nov 05 14:41:19 raspberrypi gunicorn[453]: [2020-11-05 14:41:19 -0500] [623] [INFO] Worker exiting (pid: 623)
Nov 05 14:41:20 raspberrypi gunicorn[453]: [2020-11-05 14:41:20 -0500] [453] [INFO] Shutting down: Master
Nov 05 14:41:20 raspberrypi gunicorn[453]: [2020-11-05 14:41:20 -0500] [453] [INFO] Reason: Worker failed to boot.
Nov 05 14:41:20 raspberrypi systemd[1]: mycodoflask.service: Main process exited, code=exited, status=3/NOTIMPLEMENTED
Nov 05 14:41:20 raspberrypi systemd[1]: mycodoflask.service: Failed with result 'exit-code'.

rbultman, 2020-11-05:
New update - it's working.

I think the issue was a bad input_name_unique.  I think characters I had used originally cause problems with mycodo.  I changed that and added some other code base on the examples.  My custom input is now working.  The new code is included here.
[grove_temp_humidity.zip](https://github.com/kizniche/Mycodo/files/5497539/grove_temp_humidity.zip)

closed 2020-11-05 by rbultman