kizniche/Mycodo #879: Conditional Function - Condition Check Issue

not5, 2020-11-12:
# Describe the problem/bug

Running a conditional Function that checks if a gpio pin is on our off is returning an error after upgrading to 8.8.8. was working previous to upgrade.

### Versions:

 - Mycodo Version: 8.8.8
 - Raspberry Pi Version: 3b
 - Raspbian OS Version: Buster desktop

### Debug

ERROR - mycodo.outputs.on_off_gpio_1208fb4a - Status check error: 'db4b7106-57b2-4eae-becb-6793c339bfb4'

kizniche, 2020-11-12:
It's impossible to diagnose without your code and particular conditional configuration options.

not5, 2020-11-12:
Here's the code and configs:
`# Check status of COLD
measurement = self.condition("{5f9d1584}")

if measurement is not None:  # If a measurement exists

    if measurement == "on":  # If the HVAC is on
        self.run_action("{868e3915}", message=self.message)  # Turn on FAN

    elif measurement == "off":  # Else If the HVAC is off
        self.run_action("{23775ace}", message=self.message)  # Turn off FAN`

[image]

kizniche, 2020-11-12:
Thanks. I'll test with an output state condition tomorrow.

not5, 2020-11-12:
Thanks. I think it has to do with adding channels to outputs that didn't have channels before.

kizniche, 2020-11-12:
Yes. That was an enormous amount of code change. There were bound to be parts I missed refactoring.

kizniche, 2020-11-13:
Just pushed a fix that will be available in the next release.

closed 2021-03-09 by kizniche