kizniche/Mycodo #887: No Internet Connection - error when attempting an upgrade

kpslc, 2020-11-19:
### Attempted the upgrade and the "no internet connection" displays

I am remotely logged into the system. After checking various items, decided to do the upgrade from 8.8.7 8.8.8.

After clicking on the "Upgrade" menu option the following error displays:

"No internet connection detected. To upgrade Mycodo automatically, you will need an internet connection. Refresh the page when one is connected."

This is interesting as I'm logged in remotely.

### Versions:

 - Mycodo Version:8.8.7
 - Raspberry Pi Version: 4
Mycodo Version: 8.8.7 
Python Version: 3.7.3 (default, Jul 25 2020, 13:03:44) [GCC 8.3.0] 
Database Version: 66e27f22b15a
Daemon Status: Running 
Daemon Process ID: 795 
Daemon RAM Usage: 59.484 MB 
Daemon Virtualenv: Yes 
Frontend Process ID: 480 
Frontend RAM Usage: 110.452 MB 
Frontend Virtualenv: Yes

Kernel Information: uname -a
Linux Hydroponics 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Expected behavior

A clear and concise description of what you expected to happen.

### Screenshots

If applicable, add screenshots to help explain your problem.

### Additional context

Is there anything that should be added to make it easier to address this issue?

kpslc, 2020-11-21:
Any idea on what can be done to get it to upgrade? 
There is an internet connection from the Raspberry Pi. I can login to it remotely. Today I visited the greenhouse and was able to visit external sites from the Raspberry. I also tried to do the upgrade locally. No luck.

kizniche, 2020-11-21:
In the General Settings you will find the section Upgrade Options that allows you to configure the host, port, and timeout for how Mycodo detects an internet connection during an upgrade check.

https://github.com/kizniche/Mycodo/blob/833e00285a0a34e599bae80e081cd42303c4198d/mycodo/mycodo_flask/routes_admin.py#L416-L420

https://github.com/kizniche/Mycodo/blob/833e00285a0a34e599bae80e081cd42303c4198d/mycodo/mycodo_flask/routes_admin.py#L416-L420

closed 2020-11-21 by kpslc