kizniche/Mycodo #891: Output GPIO not set as output when Pi / Mycodo restarts

LilianV, 2020-11-22:
Hello,

I have 4 outputs in my Mycodo, set up as "ON/OFF GPIO" type on pin 6, 13, 19 and 26.

When I restart the Pi, the outputs are not working because the GPIO's are not set as outputs until I open each of them and click "save".

Is it a normal behavior ?

 - Mycodo Version: 8.8.8
 - Raspberry Pi Version: Pi 4B Rev 1.2
 - Raspbian OS Version: Buster Lite

Thanks for help.
Lilian

kizniche, 2020-11-24:
Can you provide the output configuration options?

LilianV, 2020-11-24:
> Can you provide the output configuration options?

Hello !

All 4 outputs are configured the same:

[image]

They looks ok at startup:

[image]

But except for one of them (GPIO 6), GPIO's do not switch as outputs:

[image]

I have to re-open the output and save it (without any change) to switch GPIO:

[image]

[image]

Thanks for your help,
Lilian

LilianV, 2020-11-24:
> Can you provide the output configuration options?

Hi again,

While answering ... I have understood why... I created those GPIO's as outputs AND inputs... stupid... but GPIO 6 was the only one disabled (as input)... sorry... my bad.

Lilian

closed 2020-11-24 by LilianV

kizniche, 2020-11-24:
Glad you figured out the issue. If you need ways of querying the state of your outputs, I can give you my recommendations. There are numerous ways so I'd have to know where you need them (Conditional, custom Function, custom Input, etc.). As you found, having GPIO inputs with outputs can cause strange behaviors.