kizniche/Mycodo #893: Output Addition: MCP4728 DAC

Magnum-Pl, 2020-11-25:
Hi Kyle,

Can you add support for the MCP4728 DAC?

I'd like to use it to control mini brushless DC compressor that uses a 0-4.5V signal to regulate speed. 

I've attached its Python library from the Adafruit GitHub repo.

[Adafruit_CircuitPython_MCP4728-master.zip](https://github.com/kizniche/Mycodo/files/5599410/Adafruit_CircuitPython_MCP4728-master.zip)

Would it be possible to add the ability to control the analog pin of an ESP8266 through MQTT?

kizniche, 2020-11-25:
I'll check out the library and see if I can come up with an Input module.

There's already an MQTT output module, so you would have to flash your ESP with an MQTT client that can subscribe to your server. The Mosquito MQTT server runs great on the Pi. I haven't worked with the ESP mush, and haven't done any work with MQTT on it, but I'm sure there are a lot of resources out there for it.

Magnum-Pl, 2020-11-25:
Thanks for the quick response. 

I was able to get the MQTT output working with the ESP and it works great to turn a pin on and off. Do you think it would be possible to use MQTT to vary the voltage on an analog pin, like a wireless DAC?

kizniche, 2020-11-25:
I think I can come up with an output to do that. It may take some reworking of the output system, but I think that would be a good step toward standardizing outputs that accept values (in addition to those that accept duty cycle, duration, and volume).

Magnum-Pl, 2020-11-27:
It'll be interesting to experiment with this kind of output. The mini water chiller I'm working on now has a compressor and an electronic expansion valve that's controlled by a stepper motor. Each is being controlled by a 0-5v signal from a potentiometer but I'd like to use a PID instead. Im using an Arduino to control the stepper driver board. Is there any way you could add an output to control a stepper motor with Mycodo?

kizniche, 2020-11-27:
#857 is where I'm going to start with stepper motor support, I just haven't gotten around to it.

Magnum-Pl, 2020-11-27:
Sounds interesting. I printed a couple peristaltic pump heads for Nema 17 and 24 steppers last year as part of an idea for an automated dosing system. Let me know if you need any help testing for the steppers, DAC or MQTT output. I have unipolar and bipolar steppers/drivers.

kizniche, 2020-12-11:
I just added support for DACs and the MCP4728. I don't have this chip, so could you upgrade to master and test if it works? This is a basic implementation, so if it works on the output page by manually changing the voltage using the set voltage input and button, then the module works. I still need to go though the system and add voltage as an output. I might refactor a good portion of the system to make it easier to create new outputs that accept values (configured in the output module file) that never before existed in Mycodo (such as voltage in this case).

Magnum-Pl, 2020-12-12:
I upgraded to master, but wasn't able to add the new output. I keep getting stuck trying to install the dependencies. I went through the install process and get the "All dependencies have been installed" message, but when I go back to add the output again I get sent back to the install dependencies page. I found Issue #804 with the same problem. I tried manually installing Adafruit-circuitpython-mcp4728 and Adafruit-extended-bus, but I still get sent to the install dependencies page.

kizniche, 2020-12-12:
It worked fine for me. The frontend should restart in order to recognize the new python libraries. Try manually restarting the frontend from the config menu.

kizniche, 2020-12-12:
Did you install this library previously on this Pi?

Magnum-Pl, 2020-12-12:
I tried restarting the frontend, but that didn't help. There was an error installing pyusb, so I manually installed it and was able to install the output. 

Theres no option to turn the output on, only an off button. I tried sending a voltage from the channel options and I get this error in the log. I get the same error for any voltage. 

`ERROR - mycodo.outputs.mcp4728_93d039ab - Invalid parameters: State: on, Type: volt, Amount: 1.0`

kizniche, 2020-12-12:
I already fixed that issue, but I guess I somehow reverted the code. I just committed the fix for that, so upgrade to master and give it another go.

Magnum-Pl, 2020-12-12:
Im now able to turn channels B, C & D on and off by sending a voltage, but no matter what voltage I send, only 2.05V is measured. Changing VREF or switching between internal or VDD VREF doesn't have any effect. Channel A still gives me the invalid parameters error. 

Each channel has a box to change VREF (Internal/VDD) and channel gain for each of the 4 channels. Each channel should only have the option to change those values for its own channel. 

[image]

kizniche, 2020-12-12:
okay I see some more issues. I'll have a fix in a moment.

kizniche, 2020-12-12:
>Each channel has a box to change VREF (Internal/VDD) and channel gain for each of the 4 channels. Each channel should only have the option to change those values for its own channel.

Can you explain what this means? I'm confused by how it's stated.

kizniche, 2020-12-12:
Just pushed some fixes.

Magnum-Pl, 2020-12-12:
The 8 boxes in the picture to the right of VREF (volts), that switch between internal or VDD VREF and the gain from 1X-2X, are shown in each of the 4 channel options. Seems like each channel option should only have the boxes to switch between internal or VDD VREF and gain for that specific channel.

kizniche, 2020-12-12:
I still don't understand the issue. Each channel can have the VREF and Gain set.

kizniche, 2020-12-12:
Oh, I see. These are the options shown for each channel. That is wrong. Let me change that.

Magnum-Pl, 2020-12-12:
Im getting this error now

`ERROR - mycodo.daemon - Could not turn output on: unsupported operand type(s) for /: 'float' and 'NoneType'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 738, in output_on
    trigger_conditionals=trigger_conditionals)
  File "/var/mycodo-root/mycodo/controllers/controller_output.py", line 286, in output_on_off
    trigger_conditionals=trigger_conditionals)
  File "/var/mycodo-root/mycodo/outputs/base_output.py", line 258, in output_on_off
    output_channel=output_channel)
  File "/home/pi/Mycodo/mycodo/outputs/mcp4728.py", line 286, in output_switch
    self.channel[output_channel].value = int(65535 * (amount / self.vref))
TypeError: unsupported operand type(s) for /: 'float' and 'NoneType'`

kizniche, 2020-12-12:
Just pushed an update. I would delete your current output and add it again just to make sure there's nothing bad lingering from the old output configuration.

Magnum-Pl, 2020-12-12:
What command can I use to update to the current master? I've just been deleting Mycodo, downloading the new master and reinstalling. 

I'm still getting that last error when I try to send a voltage after updating.

kizniche, 2020-12-13:
>What command can I use to update to the current master?

https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch

I just pushed another fix that should make it finally work.

Magnum-Pl, 2020-12-13:
It works great! Im able to send a voltage up to 5V on each of the inputs.

kizniche, 2020-12-13:
Great!

I like the idea of an MQTT output that sends a value. I'll see if I can put something together for that soon.

Magnum-Pl, 2020-12-13:
Yeah, I think we can find some interesting ways to use that kind of output. 

Are you going to add a new function and action to use this new voltage output?

kizniche, 2020-12-13:
Sure. If you find any place within Mycodo that you would expect to see the output but it doesn't show, you can let me know in this issue and I'll see about adding it. I'm still working through the idea of improving the output system to allow any type of numerical output, not just the hard-coded set (pwm, volume, volts).

kizniche, 2020-12-13:
This will also allow the stepper motor output to be easily integrated once I get to working on that.

kizniche, 2020-12-13:
Actually, I think I'll try to implement the generic value output system tonight. The DAC output will likely change to use it. I'll let you know if you need to change anything about your install of master, if you care to keep that system going, and if I push any code.

Magnum-Pl, 2020-12-13:
Ok. Id like to be able to vary the output voltage from 0-5V based on an input from a temperature sensor. Im trying to do that with a PID now. I can add the output, but I think its trying  to send a value that's higher than what's expected. I was trying to do this earlier and it was sending 6-7V. Will it be possible to use a PID to vary the voltage this way?

kizniche, 2020-12-13:
I'm currently working on the PID controller working with the new output. I'll let you know my progress and when I have code to test.

kizniche, 2020-12-13:
>Can I use a PID to vary the voltage this way?

Yes, you should be able to. You'll need to set the min and max value for each lower and raise, and tune the PID so the output is also near or within this range.

kizniche, 2020-12-13:
I just made a big commit that refactors the volt output to a generic value output. You may have to delete your output and readd it for it to work, after updating. Let me know if you experience any issues. I'm going to continue looking for issues and also add a function action.

kizniche, 2020-12-13:
I'd also delete and readd the PID controller.

Magnum-Pl, 2020-12-13:
OK. I'll try it out. 

I have an unrelated question about PIDs. The Nest thermostat in my garage recently died and I set up a MQTT relay and conditional function to replace it until the thermostat I ordered gets here. The simple conditional function I set up works great but I originally tried to use a PID and was unsuccessful. The PID wants to use short on durations, but my furnace needs to be on for 60 seconds before the blower even starts. If I set the minimum on duration, anything less than what I set is ignored. Is there any way to force a minimum on duration of >60 seconds and >600 and let the PID decide how long the durations should be to keep the set point?

kizniche, 2020-12-13:
You will need to set the PID Period to greater than the minimum on period you want. For instance, if you want 60 < setpoint < 600, then I would set the PID Period to 600. You will need to tune with the integrator min/max so the PID output doesn't rise above 600, and the gains of course for responsiveness and accuracy. Does that answer your question?

kizniche, 2020-12-13:
For example, if you have a period of 600, and a Kp of 1, then an integrator max of 600 will stop the PID output at 600. A Kp of 0.5 would require an integrator max of 1200 for a max PID output of 600.

kizniche, 2020-12-13:
For a full example, if you're trying to heat the garage and the current temperature is 10 and your target temperature is 25, that's an error of 15. With the following tuning:

- Kp = 20
- Ki = 2
- kd = 0
- Integrator Max = 150

This would allow an output from 0 to 600 when the error is 0. When the error is 15, the max output can be: (P: 20 * 15 = 300) + (I: 2 * 150 = 300) + (D: 0) = 600

kizniche, 2020-12-13:
Also, in the last code I pushed, there's a new MQTT Value Output that allows values to be sent via MQTT. It's currently untested.

Magnum-Pl, 2020-12-13:
Thats awesome. Thanks. Im still learning how the PID works, mostly by trial and error. I'll try this and see how it compares to the conditional I've been running for the past few days. Before I got the conditional running properly I had a timer running the furnace for a few minutes every hour, but that wasn't very efficient. 

I'll try the new MQTT output too. I should be able to find a script for the ESP8266 to receive it.

kizniche, 2020-12-13:
Tuning PIDs can be tough, especially for very large systems that take a long time to change, like temperature in a large airspace, because you have to wait a long time to run a new tuning. I suppose you could just open the garage and let out all the heat to get it cold to be able to test a new tuning, but that sounds like a waste of a lot of energy/money. The general tuning procedure I use is as follows (and is documented in the manual):

With your Min and Max on and off times properly set as to not damage your heater, start the first tuning with a low Kp and Ki and Kd disabled (set to 0). Let the PID run to see how the system responds and if it stabilizes. With a low Kp you should see an oscillation and a rather large error (the temperature constantly stays lower than the setpoint because the heater doesn't turn on long enough). Next, increase Kp and run the PID again until it stabilizes. You should see that the average temperature is higher than the previous tuning, but an oscillation probably still occurs. Keep increasing Kp, but at some point things may not work well at all, and you'll see strange things like very large oscillations above and below the setpoint. At this point you need to back the Kp back off maybe 20 - 30 % to where it was stablish but with a persistent oscillation below the setpoint.

Now, start to slowly increase the Ki, starting with a very low value. If your Kp is 10, then start the Ki at 0.5 or 1. If Kp is 5, start Ki at 0.1 or 0.5. The Ki will start to increase the output when a persistent error occurs. This will start to slowly increase the total PID output every cycle, decreasing the error. If you see that the PID output is maxed out and no longer rising, yet there is still an error with a temperature lower than the setpoint, then increase the Ki a little more. You might also increase the Kp a little more as well, but do it in small steps so you can learn how the system responds to it.

Once you've tuned the Kp and Ki as best you can, yo may notice there's still a small oscillation. This can be diminished by now adding Kd. This should also be added in small increments. This will add a value of the opposite sign if the temperature is going to shoot past the setpoint. For instance, if the temperature is rising at a steep angle, the Kd will add a negative value to lower the PID output and cause less heat to be generated.

Hope this helps. I can't provide any real values, as this all depends on your airspace, the heater, the temperature, the insulation of the garage, and many more factors. Best of luck with the tuning!

Magnum-Pl, 2020-12-13:
That helps a lot. I unsuccessfully played around with the tuning before, but I think I have a better idea of what to do now.

kizniche, 2020-12-13:
Just FYI I pushed code that adds the stepper motor output (ref https://github.com/kizniche/Mycodo/issues/857#issuecomment-743957873).

Magnum-Pl, 2020-12-13:
I saw that. I'll try it out. I don't have that exact driver, but I think it should work with the A4998 driver I have.

kizniche, 2020-12-13:
Just added start and shutdown options to select either the saved state or a specified value.