kizniche/Mycodo #898: pH sensor not reading

muhammad-khan1011, 2020-12-07:
good day.
im using a raspi 4, atlas scientific pH and EC probes with a tentacle mini and the ezo circuits for them.
the version of Mycodo is 8.8.8 
the pH reads fine but no data from the EC probe
however the EC probe accepts calibration.

this is the error 
Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last): 

would really appreciate some assistance

kizniche, 2020-12-07:
Hi. The error you posted occurs after the error that indicates why the input crashed. You'll need to review the log and find the first input error following the input activation log line.

muhammad-khan1011, 2020-12-07:
hi thanks for your reply here is the first input error 
ValueError: could not convert string to float: '466.4,252,0.23,1.000'
2020-12-07 01:41:17,862 - INFO - mycodo.controllers.controller_input_ab2336db - Activated in 2209.9 ms
2020-12-07 01:41:17,863 - ERROR - mycodo.controllers.controller_input_ab2336db - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):

can you assist?

kizniche, 2020-12-07:
I don't see the timestamp or log level/ID info for that error. Please paste the log exactly as it appears without redaction. Also, is there should also be a firmware line in the format "Atlas Scientific Board: {brd}, Rev: {rev}, Firmware: {fw}" upon activation. Are you communicating via I2C, UART, or FTDI?

closed 2021-02-02 by kizniche