kizniche/Mycodo #900: Multiple SPI devices sharing same pins

not5, 2020-12-10:
### Describe the problem/bug

Unable to share MISO, MOSI, and Clock pins between different SPI sensors.

### Versions:

 - Mycodo Version: 8.8.8
 - Raspberry Pi Version: 3B+
 - Raspbian OS Version: Buster Desktop

### Reproducibility

Use multiple SPI sensors with same MISO, MOSI, and Clock Pin, but different CS pins

### Expected behavior

Expect sensors to work and record measurements, but after activating a second SPI sensor, the original no longer records measurements.

### Debug output

`2020-12-10 13:33:56,113 - ERROR - mycodo.inputs.max31865_c59204d3 - InputModule raised an exception when taking a reading: The GPIO channel has not been set up as an OUTPUT
2020-12-10 13:33:57,141 - ERROR - mycodo.inputs.max31865_c59204d3 - InputModule raised an exception when taking a reading: The GPIO channel has not been set up as an OUTPUT
2020-12-10 13:33:58,172 - ERROR - mycodo.inputs.max31865_c59204d3 - InputModule raised an exception when taking a reading: The GPIO channel has not been set up as an OUTPUT
2020-12-10 13:33:58,174 - ERROR - mycodo.controllers.controller_input_c59204d3 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.`

kizniche, 2020-12-11:
What sensor? Have you successfully accomplished measuring from multiple sensors on the same pins with the library used by the specific Input module in Mycodo?

not5, 2020-12-11:
It's the MAX31865 sensor. I have not tried the process outside of Mycodo; however, Googling the SPI interface and RPi indicates that at least two sensors should be able to share the same pins (excluding the CS pin, which must be separate for each sensor). Others have uses the library and it's worked successfully: https://www.raspberrypi.org/forums/viewtopic.php?t=236174

Also, separately activated, both work, so I think it's a Mycodo issue locking out the pins from the other sensor being activated.

kizniche, 2020-12-11:
Each Input module in Mycodo relies on 3rd party libraries (with a few exceptions) for sensor measurements. If there is an issue, there's a high chance it's an issue with that library. I would test your 2 sensor setup with whatever library is being used outside Mycodo. If it's the library's limitation, and you can find another library that doesn't have the same limitation, then I would suggest creating another input for that sensor that uses the new library.

kizniche, 2020-12-11:
>Others have used the library and it's worked successfully, eg https://www.raspberrypi.org/forums/viewtopic.php?t=236174

The library used in the Mycodo Input and on that forum are not the same.

not5, 2020-12-11:
Ok, i'll try using the library outside of Mycodo and report back.

kizniche, 2020-12-12:
If you can provide example code for specifying GPIO pins (BCM numbering) for the SPI to get the circuitpython library to work (https://github.com/adafruit/Adafruit_CircuitPython_MAX31865), I'll look into creating a new Input. Currently, I don't have any SPI circuitpython inputs to base this new input on, and I don't have this chip to test one, let alone two sensors connected at the same time.

not5, 2020-12-12:
I actually started working on trying to use the circuit python library this morning. I'm adapting the board.get_pin() function to use BCM for the CS pin. Hope to have something to test tonight.

not5, 2020-12-13:
Kyle, sent you an email. I got everything working except I can't get past the dependency requirement being installed... it seemingly wants to keep installing itself. Also, I thought there was a board.get_pin() function in the board module for circuitpython, but I was wrong. I used a very inelegant array to convert from BCM (GPIO numbering) to what circuitpython labels the pins in their board module.

kizniche, 2020-12-13:
Nice! I just committed the code. I fixed an issue with the dependency tuple... The second item needs to be the module name that's imported (since this can differ from the module name to install, the 3rd item). Other than that, I did some minor tweaking. Also, I think the list to convert to BCM works just fine. However, how would one change the SPI pins if they wanted to use one of the other SPI buses?

not5, 2020-12-13:
I think because it uses the hardware SPI of the pi, you can't change the SPI pins. If it were software SPI, then it wouldn't matter.

kizniche, 2020-12-14:
There is a board.MOSI_1 in addition to board.MOSI (as well as a "_1" for MISO and SCK). Should we add a dropdown to select between using each set of pins?

kizniche, 2020-12-14:
Additionally, [adafruit_blinka.board.raspberrypi.raspi_cm](https://github.com/adafruit/Adafruit_Blinka/blob/master/src/adafruit_blinka/board/raspberrypi/raspi_cm.py) has a third SPI ("_2").

not5, 2020-12-14:
There are two hardware SPI buses that the Pi has. I think the second bus is only really used when you have a device that dominates the other bus, like an LCD.

kizniche, 2020-12-14:
Can't they also run at different baud rates to accommodate devices at different speeds?

not5, 2020-12-14:
Yes, they can its just a serial protocol that uses 4 wires. You're starting to get outside of my knowledge sphere regarding the Pi 😋

closed 2020-12-15 by not5