kizniche/Mycodo #901: Set the desired PH and EC level thru Widget

chamroeunbm, 2020-12-11:
**Is your feature request related to a problem? Please describe.**
The hydroponics ec/ph regulation is a static value, for example PH down is working when PH is below the set static value

**Describe the solution you'd like**
Be able to set these(EC/PH) thru the widget

**Describe alternatives you've considered**
I need your help on making these communication blocks(Widget, Functions) to work together in an efficient way and follow your design pattern.

**Additional context**

kizniche, 2020-12-22:
I've been thinking about how to do this, which is why I haven't replied yet. I think I have a good solution. I just pushed a fix that adds the functions get_custom_option() and set_custom_option() to Conditionals. This allows values to be saved and retrieved from the database from Conditional Statement Python code. For example:

```python
self.logger("Conditional just started: {}".format(self.get_custom_option("my_value")))
set_custom_option("my_value", 20)
self.logger("Conditional already started: {}".format(self.get_custom_option("my_value")))
```

If this is a new conditional that is ran for the first time, the output will be:

```
Conditional just started: None
Conditional already started: 20
```

Henceforth, when the Conditional executes, it will return:

```
Conditional just started: 20
Conditional already started: 20
```

This allows you to set a pH value, for instance, which can then be modified externally by other processes:

```python
if self.get_custom_option("pH") == None:
    # This code only executes once, when the Conditional is activated for the first time
    set_custom_option("my_value", 6.5)

ph_value = self.get_custom_option("pH")

self.logger("pH value is {}".format(ph_value))
# Code to do stuff with ph_value goes here
```

This will set the pH value in the database to 6.5 the first time the Conditional is activated. From then on, every time the Conditional runs, it will pull the value from the database. You only now need to edit that value in the database and the next time the Conditional runs, it will use the new value that you set. From here, all you need to do is create a Widget that changes this value in the database and the pH adjustment Conditional will start using that new value the next time is runs.

To test the latest code on the master branch, you'll need to [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch).

chamroeunbm, 2020-12-29:
thanks. set/get_custom_options directly to db is a great feature

closed 2021-01-04 by chamroeunbm