kizniche/Mycodo #902: MCP3008 error

grux77, 2020-12-12:
Hello Kyle!
I have a problem with A/D converter MCP3008.
For testing I connected a potentiometer resistor 10kOhm to Chanal0.
Hardware and wiring are ok because the simpletest.py from Adafruit is successful.
[image]

[image]
RPi B & 4
OS: Buster light
Mycodo Version: 8.8.8
Python Version: 3.7.3 (default, Jul 25 2020, 13:03:44) [GCC 8.3.0]
Database Version: 66e27f22b15a
Daemon Status: Running
Frontend Process ID: 342
Frontend RAM Usage: 54.168 MB
Frontend Virtualenv: Yes

Here is the log:
2020-12-12 13:56:09,121 - ERROR - mycodo.controllers.controller_input_6cca7a09 - initialize_variables() Exception: 'Input' object has no attribute 'scale_from_max'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 72, in run
    self.initialize_variables()
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 338, in initialize_variables
    self.measure_input = input_loaded.InputModule(self.input_dev)
  File "/home/pi/Mycodo/mycodo/inputs/mcp3008.py", line 66, in __init__
    self.initialize_input()
  File "/home/pi/Mycodo/mycodo/inputs/mcp3008.py", line 71, in initialize_input
    self.scale_from_max = self.input_dev.scale_from_max
AttributeError: 'Input' object has no attribute 'scale_from_max'
2020-12-12 13:56:09,143 - INFO - mycodo.controllers.controller_input_6cca7a09 - Activated in 3243.3 ms
2020-12-12 13:56:09,146 - ERROR - mycodo.controllers.controller_input_6cca7a09 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 391, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'

kizniche, 2020-12-12:
I just pushed a potential fix. Could you please [upgrading to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) and test if the fix worked and the proper voltage is being measured/calculated?

grux77, 2020-12-12:
Yes it works!  You're as fast as lightning again!
But I can't save 'Invert Scale'.

[image]

kizniche, 2020-12-12:
It appears to have been saving properly, just not displaying the saved value. Just pushed a fix. Thanks for all the bug reports. Can you verify the measured/calculated voltage is accurate?

grux77, 2020-12-12:
Yes, the voltage is measured correctly.

One more thing: why is 'Clock Pin' listed twice? A blemish?
[image]

kizniche, 2020-12-12:
>why is 'Clock Pin' listed twice?

That should have been fixed 5 hours ago (f35033cea14a96aa3f393dce7baeb92c8980d899).

grux77, 2020-12-12:
Oh sorry! I didn't even notice. Embarrassing!

closed 2021-03-09 by kizniche