kizniche/Mycodo #903: MiFlora no values in Mycodo

rowkois, 2020-12-14:
### Describe the problem/bug

Adding MiFlora sensor is not giving any measure results. 
Result of hcitool lescan: C4:7C:8D:XX:XX:XX Flower care
Information from Logfile
2020-12-14 16:16:13,403 - ERROR - mycodo.controllers.controller_input_2a4980bb - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.

### Versions:

 - Mycodo Version: 8.8.8
 - Raspberry Pi Version: 2B 2014
 - Raspbian OS Version: 2020-12-02-raspios-buster-armhf-lite.zip

### Reproducibility

Add Sensor from dropdown list.
Add correct Bluetooth interface according to hciconfig (in my case: 0)
Add Mac of sensor, according to hcitool lescan
safe settings
activate sensor interface
check resulsts from data site

### Expected behavior

informations about ilumination, batterie, conductivity, etc.

### Screenshots

If applicable, add screenshots to help explain your problem.

### Additional context

Is there anything that should be added to make it easier to address this issue?

kizniche, 2020-12-14:
Have you been able to successfully acquire measurements from a device other than Mycodo (such as a phone app or other software)?

kizniche, 2020-12-14:
Have you run the Input with "Log Level: Debug" enabled to see if there is additional log output that can help investigate the issue?

rowkois, 2020-12-15:
Yes, The app gives me the values.

The log informations are active, the following I have found:
Web:
ec 15 05:49:20 raspberrypi gunicorn[8513]: 2020-12-15 05:49:20,261 TEST03: MIFLORA
Dec 15 05:49:20 raspberrypi gunicorn[8513]: 2020-12-15 05:49:20,262 TEST02: MIFLORA
Dec 15 05:49:20 raspberrypi gunicorn[8513]: 2020-12-15 05:49:20,410 TEST01: miflora, ModuleSpec(name='miflora', loader=<_frozen_importlib_external.SourceFileLoader object at 0x723ec8b0>, origin='/home/pi/Mycodo/env/lib/python3.7/site-packages/miflora/__init__.py', submodule_search_locations=['/home/pi/Mycodo/env/lib/python3.7/site-packages/miflora'])
Dec 15 05:49:20 raspberrypi gunicorn[8513]: 2020-12-15 05:49:20,414 TEST01: btlewrap, ModuleSpec(name='btlewrap', loader=<_frozen_importlib_external.SourceFileLoader object at 0x723ecad0>, origin='/home/pi/Mycodo/env/lib/python3.7/site-packages/btlewrap/__init__.py', submodule_search_locations=['/home/pi/Mycodo/env/lib/python3.7/site-packages/btlewrap'])
Dec 15 05:49:20 raspberrypi gunicorn[8513]: 2020-12-15 05:49:20,417 TEST01: bluepy, ModuleSpec(name='bluepy', loader=<_frozen_importlib_external.SourceFileLoader object at 0x723ec9d0>, origin='/home/pi/Mycodo/env/lib/python3.7/site-packages/bluepy/__init__.py', submodule_search_locations=['/home/pi/Mycodo/env/lib/python3.7/site-packages/bluepy'])

Daemon:
2020-12-15 05:49:58,433 - ERROR - mycodo.controllers.controller_input_2a4980bb - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-12-15 05:50:00,348 - INFO - mycodo.controllers.controller_input_2a4980bb - Deactivated in 39.1 ms
2020-12-15 05:54:51,994 - INFO - mycodo.controllers.controller_input_d98f7e53 - Activated in 703.0 ms
2020-12-15 05:55:02,512 - INFO - mycodo.controllers.controller_input_2a4980bb - Activated in 1098.9 ms
2020-12-15 05:55:21,783 - ERROR - mycodo.controllers.controller_input_d98f7e53 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-12-15 05:56:06,813 - ERROR - mycodo.controllers.controller_input_d98f7e53 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-12-15 05:56:51,827 - ERROR - mycodo.controllers.controller_input_d98f7e53 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-12-15 05:57:19,708 - INFO - mycodo.controllers.controller_input_d98f7e53 - Deactivated in 8.3 ms
2020-12-15 05:57:32,368 - ERROR - mycodo.controllers.controller_input_2a4980bb - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-12-15 05:57:41,306 - INFO - mycodo.controllers.controller_input_d98f7e53 - Activated in 765.9 ms
2020-12-15 05:58:17,337 - ERROR - mycodo.controllers.controller_input_2a4980bb - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.

kizniche, 2020-12-16:
I don't own this device to be able to test and diagnose any issues. Did this work previously on differnet hardware or a different Mycodo version? If so, what has changed? I know @RynFlutsch has used this device in the past. @RynFlutsch, can you confirm if your MiFlora works in the latest release?

kizniche, 2020-12-16:
@cohe4ko, @samsixtysix, and @fry42 also used this device in the past, perhaps they are still using it and can help determine if it's an issue with Mycodo. If any of you are still running this device, on the latest Mycodo v8.8.8, could you let us know if it's still working? Thanks

Perhaps it's the version the miflora python library installed. Try this command to upgrade, then restart:

```bash
sudo ~/Mycodo/env/bin/pip install --upgrade miflora
```

It also couldn't hurt to make sure your system is up to date with the latest firmwares and restart:

```bash
sudo apt update && sudo apt upgrade
```

kizniche, 2020-12-16:
Also, does https://github.com/basnijholt/miflora/blob/master/demo.py work outside of Mycodo? If not, then it seems it may be a configuration or hardware issue and not related to Mycodo.

rowkois, 2020-12-16:
the exact same hardware setting has worked previously, the version was 6.1.X and afterwards also, unfortunately the SD-Card has died.
Upgrades as above are done, no change after the restart.
the demo.py showed values, firmware, etc.
Thanks for the fast response

rowkois, 2020-12-25:
any new ideas regarding my problem? the mifora demo.py runs fine outside of mycodo, so I don't know what might interfere  with mycodo

kizniche, 2020-12-26:
Which backend did you select for demo.py?

kizniche, 2020-12-26:
Can you attach a screenshot of your input configuration options?

rowkois, 2020-12-26:
What do you mean by backend exactly? I am not a Software developer, so i am sorry for that questions.
I followed the how-to from this site     https://zsiti.eu/xiaomi-miflora-plant-sensor-pimatic-raspberry-pi-3/
but it also worked with this command: python3.7 demo.py --backend gatttool poll C4:7C:8D:

here the requested screenshot, I have not settings for the 'Convert Measurements'
[image]

kizniche, 2020-12-26:
The miflora input uses the bluepy backend. You can see which backend you're using with demo.py with the --backend argument you're using, and it isn't the same one used by the Mycodo Input.

https://github.com/kizniche/Mycodo/blob/0f2f2043340c6752ebb90fba5a1e4e27079cf008/mycodo/inputs/miflora_sensor.py#L80-L83

rowkois, 2020-12-26:
Ok, now I got it.
Using the bluepy backend, also provides data from the sensor
[image]

kizniche, 2020-12-26:
I don't know what to tell you. The same code that's in demo.py is what's in the Mycodo input. Since I don't own the sensor, I can't diagnose anything myself. The best I can do it add debug log lines that will appear in the daemon log if there's an exception. If you want to test the lastest code with an additional log line added, you can [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) and test.

rowkois, 2020-12-26:
ok, I upgraded to master, as suggested by you. Somehow, now Mycodo gives me the values.
It seems the problem is solved now.

kizniche, 2020-12-28:
Great. Glad to hear.

closed 2020-12-28 by rowkois