kizniche/Mycodo #906: Fresh install, Compiling mycodo_wrapper upgrade_commands warnings

mrplow, 2020-12-15:
### Describe the problem/bug

During a fresh install I receive a couple warnings. Doesn't seem to affect anything, but new users may worry.

### Versions:

 - Mycodo Version: 8.8.8
 - Raspberry Pi Version: pi-zero
 - Raspbian OS Version: Buster Lite

### Additional context

Log snippet
```
#### Compiling mycodo_wrapper
/home/pi/Mycodo/mycodo/scripts/mycodo_wrapper.c: In function ‘upgrade_commands’:
/home/pi/Mycodo/mycodo/scripts/mycodo_wrapper.c:17:5: warning: ‘strncat’ specified bound 255 equals destination size [-Wstringop-overflow=]
     strncat(full_cmd, "/upgrade_commands.sh ", sizeof(full_cmd));
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/pi/Mycodo/mycodo/scripts/mycodo_wrapper.c: In function ‘main’:
/home/pi/Mycodo/mycodo/scripts/mycodo_wrapper.c:35:4: warning: ‘strncat’ specified bound 255 equals destination size [-Wstringop-overflow=]
    strncat(restoreScript, "/upgrade_commands.sh backup-create", sizeof(restoreScript));
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

```
Full install.log https://termbin.com/8o5f6

kizniche, 2020-12-16:
That's expected. They're only warnings, which don't affect the compiled binary.

kizniche, 2020-12-16:
Just committed a fix

ebo, 2020-12-17:
Out of curiosity, what was the fix?  I was going to suggest a couple of fixes that would either limit the max size of the string or otherwise clip the result.  Not enforcing sting limits can allow buffer overrun attacks.

kizniche, 2021-01-24:
See cfb6e3f42358b1a2d8955b10285f052b5eec13fe

closed 2021-03-09 by kizniche