kizniche/Mycodo #908: install on RPi3 starts installing every version of pip available.

ebo, 2020-12-17:
I have an RPi3 that I use to experiment with.  Last night I updated the system and tried to install Mycodo using the supplied install instructions and end up installing about 20+ different versions of pip before I killed the install.

Has anyone ever seen this behavior before?  I should have some time this weekend to dive into the issue.  I thought I would ask the community to prep before setting up a debugging session.  

Thanks

kizniche, 2020-12-19:
>Last night I updated the system and tried to install Mycodo using the supplied install instructions and end up installing about 20+ different versions of pip before I killed the install.

That sounds very strange. Can you attach the setup.log that's generated in the install directory?

ebo, 2020-12-19:
Here it is: [setup.log](https://github.com/kizniche/Mycodo/files/5719111/setup.log)

I also went back and tried it again this morning just to make sure that I did what I thought I did:

* remove old version
* run "curl -L https://kizniche.github.io/Mycodo/install | bash" -- as a note, I also alternatively tried running the setup script independently
* then I killed it after it appeared to install the 12'th or 13'th verion -- I assumed that from the "using cached pip-*" messages.

I decided to run it again, and just let it do its thing.  It took 26 minutes, but finished.  Is this an efficiency issue, or did I misinterpret something?

As a note, one oddity of my setup is that I prefer to run off of thumb drives instead of SD Cards -- this is on a mounted setup with a touch screen.  Getting to the SD Card is a bit of a hassle, so I typically install and run off of thumb drives.

Let me know if there is anything else you would like.  I can also send the setup.log of the finished one, but it is very long, and basically is more of the same of the one up top.  As a note, it is likely that there is some trivial issue with the install script for pip, and that that there are not overall issues other than it downloading pip, and its dependencies, over and over for each version.  So, this is likely just an efficiency thing.

kizniche, 2020-12-19:
I've never seen that behavior before. I just installed Raspberry Pi OS Buster Lite and Mycodo on a RPi 3 and it only installed the latest pip version. What OS are you running?

ebo, 2020-12-19:
It has been at least a year since I installed this machine, but I update it each time I turn it on to do anything serious, and a few extra times for good measure.  So, this is the information I got from various files on the system...

From /etc/debian_version I get: 10.7, and according to the website, that is the latest update from 2020/12/5

From /boot/issue, which is the same as /etc/rpi-issue, I get:  Raspberry Pi reference 2019-07-10.  This is believable for this USBLive thumb drive install.  Oddly enough this is the thumbdrive that I tried to install Zoom on when my Master Gardner class went from in person to zoom, which should have been around 2020/04.  Anyway, after awhile I gave up on zoom, so this is not an install on top of a fresh image.

I am working on finding my notes to make more/new USBLive images, and will try setting up a clean Buster, and Gentoo again.

ebo, 2020-12-19:
also, I am pretty sure I did not install it on a Buster Lite image, but the full Buster.  I think I have enough thumb drives to set up installs for both the latest Buster Lite and full on Buster.  I should also have at least one to install Gentoo on as well.

ebo, 2020-12-29:
Forgot to close this issue of the weekend -- I started with a clean version of Buster (full desktop), and it installed without errors.  I am closing this as it looks like it was something weird with my setup.  If I ever figure out why it did this I will post it, but frankly I do not think it is worth anyone's time at this point.

Also to note, I got both Gentoo(64) and a clean Buster working this last weekend.  I do not have any sensors where I am sheltering, so I am going to purchase some as soon as they open after the holiday.  I have some unrelated issues I will post elsewhere.  Thanks for the feedback.

closed 2020-12-29 by ebo