kizniche/Mycodo #909: Error 502: Bad Gateway Error

electricspaceman, 2020-12-17:
Hi,

Receiving this error. I am using the master branch.

[image]

kizniche, 2020-12-17:
The line number isn't what's found on the master branch, so it's probably something you did to the code.

not5, 2020-12-17:
As @kizniche said, the error is not part of an official release-- my recommendation would be to delete the max31865_circuitpython.py input from the folder and restart. If you're messing with the github inputs, you should really be importing it as a custom input (with a unique name, of course).

kizniche, 2020-12-17:
From https://github.com/kizniche/Mycodo/issues/570#issuecomment-747459599, it looks like you commented out the only line in ```finally```, therefore it's expecting the return to be indented to that level. You should put ```pass``` in ```finally``` or remove the try/finally block all together.

closed 2020-12-18 by electricspaceman