kizniche/Mycodo #910: DS18B20 ow-shell

electricspaceman, 2020-12-19:
Hi Kyle, 

Wondering if you could please check out this log - trying to get DS18B20 working. Currently reading 0. 

`2020-12-19 14:30:33,246 - ERROR - mycodo.inputs.ds18b20_owshell_db551ee5 - 1
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/inputs/ds18b20_owshell.py", line 99, in get_measurement
    temperature = float(owread_output.decode("latin1"))
ValueError: could not convert string to float: 
2020-12-19 14:30:33,306 - ERROR - mycodo.inputs.ds18b20_owshell_db551ee5 - 1
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/inputs/ds18b20_owshell.py", line 99, in get_measurement
    temperature = float(owread_output.decode("latin1"))
ValueError: could not convert string to float: 
2020-12-19 14:30:33,314 - ERROR - mycodo.inputs.ds18b20_owshell_db551ee5 - Cannot set a value of None. Must be a float or string representing a float. Check the sensor and Input module is working correctly.
2020-12-19 14:30:47,980 - ERROR - mycodo.inputs.ds18b20_owshell_db551ee5 - 1
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/inputs/ds18b20_owshell.py", line 99, in get_measurement
    temperature = float(owread_output.decode("latin1"))`

Many thanks in advance for your time!

kizniche, 2020-12-22:
Do you have the sensor wired correctly, 1-wire enabled, and you've successfully used owshell to obtain a measurement outside of Mycodo?

electricspaceman, 2020-12-23:
Yes.

electricspaceman, 2020-12-23:
[image]

kizniche, 2021-01-06:
Can you enable debug mode, activate the input, and copy the daemon log output, specifically the line that states "Output: "?

electricspaceman, 2021-01-07:
`envcontrol - 8.8.8
01/07 09:31
Data
Setup
More
Select the number of lines to display from the end of a log

100

Daemon
Last 100 lines of /var/log/mycodo/mycodo.log:
2021-01-07 09:29:17,807 - ERROR - mycodo.inputs.ds18b20_owshell_cebfc8f7 - Cannot set a value of None. Must be a float or string representing a float. Check the sensor and Input module is working correctly.
2021-01-07 09:29:32,824 - ERROR - mycodo.inputs.ds18b20_owshell_cebfc8f7 - 1
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/inputs/ds18b20_owshell.py", line 99, in get_measurement
    temperature = float(owread_output.decode("latin1"))
ValueError: could not convert string to float: 
2021-01-07 09:29:32,870 - ERROR - mycodo.inputs.ds18b20_owshell_cebfc8f7 - 1
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/inputs/ds18b20_owshell.py", line 99, in get_measurement
    temperature = float(owread_output.decode("latin1"))
ValueError: could not convert string to float: 
2021-01-07 09:29:32,872 - ERROR - mycodo.inputs.ds18b20_owshell_cebfc8f7 - Cannot set a value of None. Must be a float or string representing a float. Check the sensor and Input module is working correctly.
2021-01-07 09:29:47,775 - ERROR - mycodo.inputs.ds18b20_owshell_cebfc8f7 - 1
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/inputs/ds18b20_owshell.py", line 99, in get_measurement
    temperature = float(owread_output.decode("latin1"))
ValueError: could not convert string to float: 
2021-01-07 09:29:47,795 - ERROR - mycodo.inputs.ds18b20_owshell_cebfc8f7 - 1
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/inputs/ds18b20_owshell.py", line 99, in get_measurement
    temperature = float(owread_output.decode("latin1"))
ValueError: could not convert string to float: 
2021-01-07 09:29:47,796 - ERROR - mycodo.inputs.ds18b20_owshell_cebfc8f7 - Cannot set a value of None. Must be a float or string representing a float. Check the sensor and Input module is working correctly.
2021-01-07 09:30:02,797 - ERROR - mycodo.inputs.ds18b20_owshell_cebfc8f7 - 1
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/inputs/ds18b20_owshell.py", line 99, in get_measurement
    temperature = float(owread_output.decode("latin1"))
ValueError: could not convert string to float: 
2021-01-07 09:30:02,816 - ERROR - mycodo.inputs.ds18b20_owshell_cebfc8f7 - 1
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/inputs/ds18b20_owshell.py", line 99, in get_measurement
    temperature = float(owread_output.decode("latin1"))
ValueError: could not convert string to float: 
2021-01-07 09:30:02,817 - ERROR - mycodo.inputs.ds18b20_owshell_cebfc8f7 - Cannot set a value of None. Must be a float or string representing a float. Check the sensor and Input module is working correctly.
2021-01-07 09:30:17,800 - ERROR - mycodo.inputs.ds18b20_owshell_cebfc8f7 - 1
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/inputs/ds18b20_owshell.py", line 99, in get_measurement
    temperature = float(owread_output.decode("latin1"))
ValueError: could not convert string to float: 
2021-01-07 09:30:17,820 - ERROR - mycodo.inputs.ds18b20_owshell_cebfc8f7 - 1
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/inputs/ds18b20_owshell.py", line 99, in get_measurement
    temperature = float(owread_output.decode("latin1"))
ValueError: could not convert string to float: 
2021-01-07 09:30:17,822 - ERROR - mycodo.inputs.ds18b20_owshell_cebfc8f7 - Cannot set a value of None. Must be a float or string representing a float. Check the sensor and Input module is working correctly.
2021-01-07 09:30:32,832 - ERROR - mycodo.inputs.ds18b20_owshell_cebfc8f7 - 1
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/inputs/ds18b20_owshell.py", line 99, in get_measurement
    temperature = float(owread_output.decode("latin1"))
ValueError: could not convert string to float: 
2021-01-07 09:30:32,866 - ERROR - mycodo.inputs.ds18b20_owshell_cebfc8f7 - 1
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/inputs/ds18b20_owshell.py", line 99, in get_measurement
    temperature = float(owread_output.decode("latin1"))
ValueError: could not convert string to float: 
2021-01-07 09:30:32,868 - ERROR - mycodo.inputs.ds18b20_owshell_cebfc8f7 - Cannot set a value of None. Must be a float or string representing a float. Check the sensor and Input module is working correctly.
2021-01-07 09:30:47,804 - ERROR - mycodo.inputs.ds18b20_owshell_cebfc8f7 - 1
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/inputs/ds18b20_owshell.py", line 99, in get_measurement
[... 16 lines cut ...]
    temperature = float(owread_output.decode("latin1"))
ValueError: could not convert string to float: 
2021-01-07 09:31:02,839 - ERROR - mycodo.inputs.ds18b20_owshell_cebfc8f7 - Cannot set a value of None. Must be a float or string representing a float. Check the sensor and Input module is working correctly.
2021-01-07 09:31:17,791 - ERROR - mycodo.inputs.ds18b20_owshell_cebfc8f7 - 1
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/inputs/ds18b20_owshell.py", line 99, in get_measurement
    temperature = float(owread_output.decode("latin1"))
ValueError: could not convert string to float: 
2021-01-07 09:31:17,855 - ERROR - mycodo.inputs.ds18b20_owshell_cebfc8f7 - 1
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/inputs/ds18b20_owshell.py", line 99, in get_measurement
[... cut ...]
2021-01-07 09:31:32,893 - ERROR - mycodo.inputs.ds18b20_owshell_cebfc8f7 - 1
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/inputs/ds18b20_owshell.py", line 99, in get_measurement
    temperature = float(owread_output.decode("latin1"))
ValueError: could not convert string to float: 
2021-01-07 09:31:32,895 - ERROR - mycodo.inputs.ds18b20_owshell_cebfc8f7 - Cannot set a value of None. Must be a float or string representing a float. Check the sensor and Input module is working correctly.`

kizniche, 2021-01-07:
I do not see any debug lines with "Output:". The debug line I previously mentioned appears to be a new addition only available on the master branch. You will need to upgrade to master and you will see the log line I mentioned.

electricspaceman, 2021-01-07:
I upgraded to master, deactivated the input, restarted, reactivated the input, here is the log. thanks for taking a look.

2021-01-07 13:23:10,962 - INFO - mycodo.daemon - Mycodo daemon v8.8.8 starting
2021-01-07 13:23:10,963 - DEBUG - mycodo.daemon - Refreshing camera settings
2021-01-07 13:23:11,005 - DEBUG - mycodo.daemon - Refreshing misc settings
2021-01-07 13:23:11,032 - DEBUG - mycodo.daemon - Anonymous statistics enabled
2021-01-07 13:23:11,057 - INFO - mycodo.pyro_daemon - Starting Pyro5 daemon
2021-01-07 13:23:11,059 - INFO - mycodo.pyro_monitor - Starting Pyro5 daemon monitor (30 min timer)
2021-01-07 13:23:11,529 - DEBUG - mycodo.daemon - Starting Output Controller
2021-01-07 13:23:11,647 - DEBUG - mycodo.controllers.controller_output - Initializing Outputs
2021-01-07 13:23:12,178 - DEBUG - mycodo.pyro_monitor - Pyro5 daemon monitor: daemon_status() response: 'alive'
2021-01-07 13:23:12,521 - INFO - mycodo.outputs.on_off_gpio_a7cb02ed - Output setup on pin 20 and turned OFF (ON=HIGH)
2021-01-07 13:23:12,522 - INFO - mycodo.outputs.on_off_gpio_a7cb02ed - Initialized in 283.7 ms
2021-01-07 13:23:12,522 - DEBUG - mycodo.controllers.controller_output - a7cb02ed (Heater) Initialized
2021-01-07 13:23:12,795 - INFO - mycodo.outputs.on_off_gpio_6d56264e - Output setup on pin 21 and turned OFF (ON=HIGH)
2021-01-07 13:23:12,796 - INFO - mycodo.outputs.on_off_gpio_6d56264e - Initialized in 270.9 ms
2021-01-07 13:23:12,797 - DEBUG - mycodo.controllers.controller_output - 6d56264e (Humidifer) Initialized
2021-01-07 13:23:13,081 - INFO - mycodo.outputs.on_off_gpio_27d0d629 - Output setup on pin 26 and turned OFF (ON=HIGH)
2021-01-07 13:23:13,082 - INFO - mycodo.outputs.on_off_gpio_27d0d629 - Initialized in 281.9 ms
2021-01-07 13:23:13,082 - DEBUG - mycodo.controllers.controller_output - 27d0d629 (Unused) Initialized
2021-01-07 13:23:13,083 - DEBUG - mycodo.controllers.controller_output - Outputs Initialized
2021-01-07 13:23:13,083 - INFO - mycodo.controllers.controller_output - Activated in 1552.3 ms
2021-01-07 13:23:13,585 - DEBUG - mycodo.daemon - Output Controller fully started
2021-01-07 13:23:13,586 - DEBUG - mycodo.daemon - Starting all activated Conditional controllers
2021-01-07 13:23:13,586 - INFO - mycodo.daemon - All activated Conditional controllers started
2021-01-07 13:23:13,587 - DEBUG - mycodo.daemon - Starting all activated Trigger controllers
2021-01-07 13:23:13,587 - INFO - mycodo.daemon - All activated Trigger controllers started
2021-01-07 13:23:13,587 - DEBUG - mycodo.daemon - Starting all activated Input controllers
2021-01-07 13:23:14,600 - INFO - mycodo.controllers.controller_input_702d1a43 - Activated in 491.7 ms
2021-01-07 13:23:15,661 - INFO - mycodo.controllers.controller_input_29d4d8ff - Activated in 513.2 ms
2021-01-07 13:23:16,708 - INFO - mycodo.controllers.controller_input_cebfc8f7 - Activated in 493.3 ms
2021-01-07 13:23:16,755 - ERROR - mycodo.inputs.ds18b20_owshell_cebfc8f7 - Obtaining measurement
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/inputs/ds18b20_owshell.py", line 100, in get_measurement
    temperature = float(owread_output.decode("latin1"))
ValueError: could not convert string to float: 
2021-01-07 13:23:16,800 - ERROR - mycodo.inputs.ds18b20_owshell_cebfc8f7 - Obtaining measurement
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/inputs/ds18b20_owshell.py", line 100, in get_measurement
    temperature = float(owread_output.decode("latin1"))
ValueError: could not convert string to float: 
2021-01-07 13:23:17,992 - INFO - mycodo.controllers.controller_input_ecb32d5d - Activated in 652.6 ms
2021-01-07 13:23:17,994 - INFO - mycodo.daemon - All activated Input controllers started
2021-01-07 13:23:17,995 - DEBUG - mycodo.daemon - Starting all activated Math controllers
2021-01-07 13:23:18,759 - INFO - mycodo.controllers.controller_math_626b5184 - Activated in 188.7 ms
2021-01-07 13:23:18,761 - INFO - mycodo.daemon - All activated Math controllers started
2021-01-07 13:23:18,761 - DEBUG - mycodo.daemon - Starting all activated PID controllers
2021-01-07 13:23:18,762 - INFO - mycodo.daemon - All activated PID controllers started
2021-01-07 13:23:18,762 - DEBUG - mycodo.daemon - Starting all activated LCD controllers
2021-01-07 13:23:18,763 - INFO - mycodo.daemon - All activated LCD controllers started
2021-01-07 13:23:18,763 - DEBUG - mycodo.daemon - Starting all activated Custom controllers
2021-01-07 13:23:18,764 - INFO - mycodo.daemon - All activated Custom controllers started
2021-01-07 13:23:18,764 - DEBUG - mycodo.daemon - Starting Widget Controller
2021-01-07 13:23:18,932 - DEBUG - mycodo.controllers.controller_widget - Initializing Widgets
2021-01-07 13:23:19,348 - DEBUG - mycodo.controllers.controller_widget - Widgets Initialized
2021-01-07 13:23:19,350 - INFO - mycodo.controllers.controller_widget - Activated in 584.7 ms
2021-01-07 13:23:19,851 - DEBUG - mycodo.daemon - Widget Controller fully started
2021-01-07 13:23:20,353 - INFO - mycodo.daemon - Mycodo daemon started in 9.390 seconds
2021-01-07 13:23:20,354 - DEBUG - mycodo.daemon - Statistics file doesn't exist, creating /home/pi/Mycodo/databases/statistics.csv
2021-01-07 13:23:20,858 - INFO - mycodo.daemon - 47.41 MB RAM in use
2021-01-07 13:23:31,658 - ERROR - mycodo.inputs.ds18b20_owshell_cebfc8f7 - Obtaining measurement
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/inputs/ds18b20_owshell.py", line 100, in get_measurement
[... cut ...]
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/inputs/ds18b20_owshell.py", line 100, in get_measurement
    temperature = float(owread_output.decode("latin1"))
ValueError: could not convert string to float: 
2021-01-07 13:32:00,299 - ERROR - mycodo.inputs.ds18b20_owshell_cebfc8f7 - Obtaining measurement
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/inputs/ds18b20_owshell.py", line 100, in get_measurement
    temperature = float(owread_output.decode("latin1"))

kizniche, 2021-01-07:
I don't think you enabled Log Level: Debug for the input. I only need the line with "Output:", so if you get anything else, it's not needed.

electricspaceman, 2021-01-07:
Right - with you now, sorry about that.

2021-01-07 13:42:28,097 - DEBUG - mycodo.inputs.ds18b20_owshell_cebfc8f7 - Output: 'b'\n''

kizniche, 2021-01-08:
So it seems there's something up with the Popen command not properly returning the temperature. This module used to work, so I'm not sure what's wrong. You can try adding the following line after command is set to make sure it's properly generating the correct bash command.

```python
self.logger.debug("Command: '{}'".format(command))
```

https://github.com/kizniche/Mycodo/blob/6ae07cf79258037d5ba1c8d7ecd55504c0cfcc1a/mycodo/inputs/ds18b20_owshell.py#L94-L100

Other than that, I'm not sure what the issue could be. I'll pull a DS18B20 out and test this soon, unless you can figure out the issue with the module in the meantime.

electricspaceman, 2021-01-09:
Hey Kyle, 
Many thanks again for taking a look. It would appear that the line you mention is already there?

[image]

electricspaceman, 2021-01-09:
`2021-01-09 12:55:05,194 - INFO - mycodo.daemon - Mycodo daemon v8.8.8 starting
2021-01-09 12:55:05,195 - DEBUG - mycodo.daemon - Refreshing camera settings
2021-01-09 12:55:05,274 - DEBUG - mycodo.daemon - Refreshing misc settings
2021-01-09 12:55:05,327 - DEBUG - mycodo.daemon - Anonymous statistics enabled
2021-01-09 12:55:05,376 - INFO - mycodo.pyro_daemon - Starting Pyro5 daemon
2021-01-09 12:55:05,379 - INFO - mycodo.pyro_monitor - Starting Pyro5 daemon monitor (30 min timer)
2021-01-09 12:55:05,816 - DEBUG - mycodo.daemon - Starting Output Controller
2021-01-09 12:55:05,938 - DEBUG - mycodo.controllers.controller_output - Initializing Outputs
2021-01-09 12:55:06,441 - DEBUG - mycodo.pyro_monitor - Pyro5 daemon monitor: daemon_status() response: 'alive'
2021-01-09 12:55:06,837 - INFO - mycodo.outputs.on_off_gpio_a7cb02ed - Output setup on pin 20 and turned OFF (ON=HIGH)
2021-01-09 12:55:06,838 - INFO - mycodo.outputs.on_off_gpio_a7cb02ed - Initialized in 310.4 ms
2021-01-09 12:55:06,838 - DEBUG - mycodo.controllers.controller_output - a7cb02ed (Heater) Initialized
2021-01-09 12:55:07,100 - INFO - mycodo.outputs.on_off_gpio_6d56264e - Output setup on pin 21 and turned OFF (ON=HIGH)
2021-01-09 12:55:07,107 - INFO - mycodo.outputs.on_off_gpio_6d56264e - Initialized in 266.0 ms
2021-01-09 12:55:07,108 - DEBUG - mycodo.controllers.controller_output - 6d56264e (Humidifer) Initialized
2021-01-09 12:55:07,387 - INFO - mycodo.outputs.on_off_gpio_27d0d629 - Output setup on pin 26 and turned OFF (ON=HIGH)
2021-01-09 12:55:07,388 - INFO - mycodo.outputs.on_off_gpio_27d0d629 - Initialized in 277.1 ms
2021-01-09 12:55:07,389 - DEBUG - mycodo.controllers.controller_output - 27d0d629 (Unused) Initialized
2021-01-09 12:55:07,389 - DEBUG - mycodo.controllers.controller_output - Outputs Initialized
2021-01-09 12:55:07,390 - INFO - mycodo.controllers.controller_output - Activated in 1569.1 ms
2021-01-09 12:55:07,891 - DEBUG - mycodo.daemon - Output Controller fully started
2021-01-09 12:55:07,892 - DEBUG - mycodo.daemon - Starting all activated Conditional controllers
2021-01-09 12:55:07,892 - INFO - mycodo.daemon - All activated Conditional controllers started
2021-01-09 12:55:07,893 - DEBUG - mycodo.daemon - Starting all activated Trigger controllers
2021-01-09 12:55:07,893 - INFO - mycodo.daemon - All activated Trigger controllers started
2021-01-09 12:55:07,894 - DEBUG - mycodo.daemon - Starting all activated Input controllers
2021-01-09 12:55:09,074 - INFO - mycodo.controllers.controller_input_702d1a43 - Activated in 678.4 ms
2021-01-09 12:55:10,056 - INFO - mycodo.controllers.controller_input_29d4d8ff - Activated in 510.7 ms
2021-01-09 12:55:11,093 - INFO - mycodo.controllers.controller_input_cebfc8f7 - Activated in 500.3 ms
2021-01-09 12:55:11,094 - DEBUG - mycodo.inputs.ds18b20_owshell_cebfc8f7 - Command: 'owread //temperature; echo'
2021-01-09 12:55:11,159 - DEBUG - mycodo.inputs.ds18b20_owshell_cebfc8f7 - Output: 'b'\n''
2021-01-09 12:55:11,160 - ERROR - mycodo.inputs.ds18b20_owshell_cebfc8f7 - Obtaining measurement
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/inputs/ds18b20_owshell.py", line 101, in get_measurement
    temperature = float(owread_output.decode("latin1"))
ValueError: could not convert string to float: 
2021-01-09 12:55:11,166 - DEBUG - mycodo.inputs.ds18b20_owshell_cebfc8f7 - Command: 'owread //temperature; echo'
2021-01-09 12:55:11,220 - DEBUG - mycodo.inputs.ds18b20_owshell_cebfc8f7 - Output: 'b'\n''
2021-01-09 12:55:11,227 - ERROR - mycodo.inputs.ds18b20_owshell_cebfc8f7 - Obtaining measurement
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/inputs/ds18b20_owshell.py", line 101, in get_measurement
    temperature = float(owread_output.decode("latin1"))
ValueError: could not convert string to float: 
2021-01-09 12:55:11,310 - DEBUG - mycodo.controllers.controller_input_cebfc8f7 - Adding measurements to InfluxDB with ID cebfc8f7-cc0e-4474-8a2b-a3e8866241c6: {}`

kizniche, 2021-01-09:
There's the issue

kizniche, 2021-01-09:
It appears you don't have the 1-wire device set.

kizniche, 2021-01-09:
There's the issue. It appears you don't have the 1-wire device set. Can you attach a screenshot of how your Input configuration appears?

electricspaceman, 2021-01-10:
Hi Kyle,

Yes that's right, but when I go to click on 1-Wire Device ID no devices appear.

[image]

kizniche, 2021-01-10:
That's something that should have been included in the first post and we could have solved this a long time ago. I'll look into it.

electricspaceman, 2021-01-11:
My apologies Kyle! I appreciate you taking a look at this.

electricspaceman, 2021-01-26:
I've been reading through the issues and I think there was a time where there were two sensor read methods spliced into one, and I think this caused other users issues when setting up, perhaps when you split the two methods into their own input modules the issue came back again?

Issue #582
Issue #746

Any help on this would be greatly appreciated.

electricspaceman, 2021-03-17:
Hey kyle sorry to bother you with this. Any word on getting it working?

kizniche, 2021-03-18:
Sorry, I haven't been able to pull a DS18B20 from one of my systems to test yet. Can you run just this code segment in its own and see what happens?

https://github.com/kizniche/Mycodo/blob/df21d491426b1647a8bd18a4abac204c3cac4de9/mycodo/mycodo_flask/routes_page.py#L2066-L2070

This should work:

```python
import re
import subprocess

devices_1wire_ow_shell = []
try:
    test_cmd = subprocess.check_output(['owdir']).splitlines()
    for each_ow in test_cmd:
        str_ow = re.sub("\ |\/|\'", "", each_ow.decode("utf-8"))  # Strip / and '
        if '.' in str_ow and len(str_ow) == 15:
            devices_1wire_ow_shell.append(str_ow)
except Exception:
    print("Error finding 1-wire devices with 'owdir'")
```

kizniche, 2022-03-23:
I know this is a little late, but I fixed some issues with this Input, mainly on the UI side detecting 1-wire devices. However, when running the input, I'm getting wildly varying measurements. I'm not sure what's wrong, but I'd recommend anyone using a DS18B20 sensor to use the Input with the w1thermsensor library over the ow-shell library. I'm going to mark this closed when the next release happens, unless there's any interest in keeping this issue alive.

closed 2022-03-28 by kizniche

kizniche, 2022-03-29:
This issue has been mentioned on **Radical DIY Forum**. There might be relevant details there:

https://forum.radicaldiy.com/t/mycodo-v8-13-0-release/820/1

smit9418, 2022-05-26:
> I know this is a little late, but I fixed some issues with this Input, mainly on the UI side detecting 1-wire devices. However, when running the input, I'm getting wildly varying measurements. I'm [...]

thank you, this just got my temp sensor working