kizniche/Mycodo #912: MQTT publish and subscribe

skippertje77, 2020-12-24:
Hi Kyle,
First post here, first note: Awesome project!

I have been playing around with mycodo for a while now, but run into a issue with an output MQTT publish and MQTT subscription(s) at the same time. 
Is has to do with the publish topic containing the / symbol. Whenever a MQTT message is published to mosquitto from mycodo with the / symbol in it, the subscribed MQTT data stops being parsed into mycodo. 
I have also created my own shell script output, as long as I do not parse an argument with a /, it works, but with a / the subsciption stops working (No Data Last 30 seconds) on next refresh until I deactivate and activate the subscription again.

To reproduce I did a fresh installation of mycodo on a different pi, with all raspbian upgrades, including a fresh Mosquitto, same issue.

Its not high priority, since there is a workaround using shell scripts (which is also way more flexible), but I guess it is good to know.

kizniche, 2020-12-24:
Can you provide detailed steps to reproduce the issue?

skippertje77, 2020-12-24:
Create working MQTT subscribe, test. Create new empty mqtt output(this publishes one paho/test/off). wait for one refresh in data/live.
Use mqtt explorer or equivalent to monitor moquitto
[image]

kizniche, 2020-12-24:
Influxdb stores numerical values. The On/Off MQTT doesn't send numerical values. You will need to use the Value MQTT output if you want such a setup to work, and this is not released and only available on the master branch. You can find how to upgrade to the master branch on the wiki.

kizniche, 2021-02-16:
Might #927 (https://kylegabriel.com/forum/general-discussion/mqtt-data-reading-problem-from-esp32) be what you were experiencing?

closed 2021-03-09 by kizniche