kizniche/Mycodo #914: Job for mycodo.service failed because the control process exited with error code.

mm1994uestc, 2020-12-27:
# Mycodo service intalling mistakes with my raspberry pi 4  
## The step installed mycodo is blow:  
1. I download the mycodo code in .zip format.   
2. unzip the master.zip source code in my raspberrpi.  
3. change directory into the install folder and run command: sudo /bin/bash ./setup.sh  
4. waitting for installing finished.  

## After waitting for installation, Here the bugs information:
[setup.log](https://github.com/kizniche/Mycodo/files/5744291/setup.log)
Follow the error message i did this blow:
1. systemctl status mycodo.service
    
[image]

2. journalctl -xe
    
[image]

3. I am trying to refresh the webpage but make nothings:  

[image]

## How can i fix the problems? Any suggestions？Thanks a lot！

kizniche, 2021-01-09:
Your setup log indicates you had an issue installing virtualenv:

```
#### Virtualenv doesn't exist. Creating...
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting virtualenv
  Downloading https://www.piwheels.org/simple/virtualenv/virtualenv-20.2.2-py2.py3-none-any.whl (5.7 MB)
ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
    virtualenv from https://www.piwheels.org/simple/virtualenv/virtualenv-20.2.2-py2.py3-none-any.whl#sha256=54b05fc737ea9c9ee9f8340f579e5da5b09fb64fd010ab5757eb90268616907c:
        Expected sha256 54b05fc737ea9c9ee9f8340f579e5da5b09fb64fd010ab5757eb90268616907c
             Got        9ad82547bcc2797da554e92d054e0a4c305bf135934d766cc46c35446655bbc1
```

mm1994uestc, 2021-01-27:
Thanks for your reply，I have installed the python3-virtualenv manually, and it's worked now.

closed 2021-01-27 by mm1994uestc