kizniche/Mycodo #915: Phidget USB relays?

ebo, 2020-12-29:
Has anyone used a USB Phidget relay similar to https://www.phidgets.com/?prodid=1020 ?  This is a simple USB controlled 4x relay.  From reading around I see a number of other relays and SDR's that are supported via the GPIO pins.  I'm curious to see if I can get the USB I/O working as it would open up suitable controllers.  If not I will look at setting up the custom I/O as mentioned in the docs.  Thanks.

kizniche, 2020-12-29:
It should be easy enough to add, since there's a python library available. There's an 8-channel output that's the most similar, which would only need minor modifications to make a new output for this device.

ebo, 2020-12-29:
I figured so, just thought I would ask before I sink time into stuff that someone had laying around, or knew there was some weird compatibility issue.

Thanks for the follow up.  My parts are due to start arriving in a few days.

closed 2021-09-20 by kizniche