kizniche/Mycodo #916: Editing: Gauge (Angular) Widget Configuration

cj-cuccaro, 2020-12-29:
I was adding additional color stops that presented me with the following...

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 639, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 603, in page_dashboard
    utils_dashboard.widget_mod(form_base, request.form)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/utils/utils_dashboard.py", line 255, in widget_mod
    mod_widget, request_form, custom_options_json_presave, json.loads(custom_options_json_postsave))
  File "/home/pi/Mycodo/mycodo/widgets/widget_gauge_angular.py", line 73, in execute_at_modification
    current_colors=sorted_colors)
  File "/home/pi/Mycodo/mycodo/widgets/widget_gauge_angular.py", line 475, in gauge_reformat_stops
    if new_stops > current_stops:
TypeError: '>' not supported between instances of 'NoneType' and 'int'

kizniche, 2020-12-29:
Please provide detailed steps to reproduce the issue.

cj-cuccaro, 2020-12-29:
1. Go to Default dashboard
    2. Add Dashboard Widget
    3. Select Gauge (Angular)
    4. Set Name – Water pH
    5. Select Measurement [Input 02 CH0] Atlas pH Ion Concentration (pH)
    6. Set Maximum = 14
    7. Click Create
    8. Click gear to enter widget configuration
    9. Change Stops to 3 – save
    10. Click gear to enter widget configuration
    11. Change Stops to 28 – save - got weird rendition of gauge
    12. Click gear to enter widget configuration
    13. Change Stops to 3 – got the following:

Error (Full Traceback):
Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 639, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 603, in page_dashboard
    utils_dashboard.widget_mod(form_base, request.form)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/utils/utils_dashboard.py", line 255, in widget_mod
    mod_widget, request_form, custom_options_json_presave, json.loads(custom_options_json_postsave))
  File "/home/pi/Mycodo/mycodo/widgets/widget_gauge_angular.py", line 73, in execute_at_modification
    current_colors=sorted_colors)
  File "/home/pi/Mycodo/mycodo/widgets/widget_gauge_angular.py", line 475, in gauge_reformat_stops
    if new_stops > current_stops:
TypeError: '>' not supported between instances of 'NoneType' and 'int'

kizniche, 2021-01-06:
I couldn't replicate your issue, but I did find other issues with both Angular and Solid gauges that I fixed in the latest commit.

cj-cuccaro, 2021-01-06:
Ok. I will test again after updating and let you know. May not be for a couple days.

Thank you,
CJ