kizniche/Mycodo #917: Error Capture Cam

rrherruzo, 2020-12-31:
Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 639, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 223, in page_camera
    camera_record('photo', mod_camera.unique_id)
  File "/home/pi/Mycodo/mycodo/devices/camera.py", line 189, in camera_record
    img_edited = img_orig.copy()
AttributeError: 'NoneType' object has no attribute 'copy'

kizniche, 2021-01-06:
This error is caused by opencv not being able to acquire an image from your camera. I've pushed a fix for the uncaught exception, however this will not fix your issue because the device you've set for opencv to use is not returning an image. You will need to diagnose why your camera isn't being detected by opencv.

rrherruzo, 2021-01-07:
I've already fixed it, updated the Raspberry and rebooted. Then I used the test commands to take a screenshot. And it began to be seen. Although not too good, I had to configure the saturation and the parameters because the image was very white.

closed 2021-01-09 by kizniche