kizniche/Mycodo #918: Bad Checksum Error when connecting MH-Z19B sensor UART - USB

latimer288, 2021-01-06:
Please DO NOT OPEN AN ISSUE:

 - If your Mycodo version is not the latest release version, please update your device before submitting your issue (unless your issue is related to not being able to upgrade). Your problem might already be solved.
 - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue.

Please complete as many of the sections below, if applicable, to provide the most information that may help with investigating your issue. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help those providing assistance by reducing the time required to help you.

### Describe the problem/bug

I am very new to programming and electronics so this is very much a learning process. i have connected the MH-Z19B via UART - USB utilizing mycodo following direction in Kyle Gabriel's hydroponic automation write-up. It appears to be functioning and the operating system is communicating with the sensor however i cannot receive data back to the Pi 4. when reviewing the logs I see the sensor communicating every 15 secs however the error reads bad checksum. Below I have included an example of the log. i believe that is all the info needed as I assume this is a small issue i am over looking.

Thanks for any assistance. 

,665 - INFO - mycodo.daemon - All activated Custom controllers started
2021-01-06 11:52:41,925 - INFO - mycodo.controllers.controller_widget - Activated in 259.3 ms
2021-01-06 11:52:42,926 - INFO - mycodo.daemon - Mycodo daemon started in 2.557 seconds
2021-01-06 11:52:42,930 - INFO - mycodo.daemon - 41.80 MB RAM in use
2021-01-06 14:02:12,572 - INFO - mycodo.controllers.controller_input_1f7a8450 - Activated in 268.1 ms
2021-01-06 14:02:12,583 - ERROR - mycodo.inputs.mh_z19b_1f7a8450 - Bad checksum
2021-01-06 14:02:19,620 - ERROR - mycodo.inputs.mh_z19b_1f7a8450 - Bad checksum
2021-01-06 14:02:34,561 - ERROR - mycodo.inputs.mh_z19b_1f7a8450 - Bad checksum
2021-01-06 14:02:49,613 - ERROR - mycodo.inputs.mh_z19b_1f7a8450 - Bad checksum
2021-01-06 14:03:04,558 - ERROR - mycodo.inputs.mh_z19b_1f7a8450 - Bad checksum
2021-01-06 14:03:19,602 - ERROR - mycodo.inputs.mh_z19b_1f7a8450 - Bad checksum
2021-01-06 14:03:34,545 - ERROR - mycodo.inputs.mh_z19b_1f7a8450 - Bad checksum
2021-01-06 14:03:49,594 - ERROR - mycodo.inputs.mh_z19b_1f7a8450 - Bad checksum
2021-01-06 14:04:04,637 - ERROR - mycodo.inputs.mh_z19b_1f7a8450 - Bad checksum
2021-01-06 14:04:19,581 - ERROR - mycodo.inputs.mh_z19b_1f7a8450 - Bad checksum
2021-01-06 14:04:34,625 - ERROR - mycodo.inputs.mh_z19b_1f7a8450 - Bad checksum
2021-01-06 14:04:49,568 - ERROR - mycodo.inputs.mh_z19b_1f7a8450 - Bad checksum
2021-01-06 14:05:04,613 - ERROR - mycodo.inputs.mh_z19b_1f7a8450 - Bad checksum
2021-01-06 14:05:19,558 - ERROR - mycodo.inputs.mh_z19b_1f7a8450 - Bad checksum
2021-01-06 14:05:34,606 - ERROR - mycodo.inputs.mh_z19b_1f7a8450 - Bad checksum
### Reproducibility

kizniche, 2021-01-06:
Make sure you have the wires connected properly and are using the correct USB device.

latimer288, 2021-01-06:
hey Kyle, I am using the exact USB device you recommended in your write up. However I had issues wiring the harness from the sensor to the USB device. It didn't seem to work when i mimicked your picture in the write up. So I wired the sensor on a breadboard to the USB device. that got a connection, however  also the error.

kizniche, 2021-01-07:
Can you write the chip model and also show how you have it wired in order to verify it's correct? There's something wrong with your configuration, and without enough information, diagnosis is impossible.

latimer288, 2021-01-07:
re wired the usb device properly and fixed the issue. thanks for the assistance and work you are doing. the write up you did of the hydroponic automation was very inspirational. looking forward to continuing this project.

kizniche, 2021-01-07:
Great! Glad to hear. Also check out the forum at kylegabriel.com/forum for any general discussions you want to start, if it's not a software issue.

closed 2021-01-07 by kizniche