kizniche/Mycodo #921: myCodo error, when I try to add a new data source.

TravisDuck999, 2021-01-16:
I'm new to this but I think this might be helping somewaht ?
I got this error when I slected :
Setup > Data 

I wanted to add a new graph if it helps, but I have added new ones already before so I was a little confused.

Traceback (most recent call last):
  File "/home/foodie/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/foodie/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/foodie/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 639, in error_router
    return original_handler(e)
  File "/home/foodie/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/foodie/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/foodie/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/foodie/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/foodie/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/home/foodie/Mycodo/mycodo/mycodo_flask/routes_page.py", line 2007, in page_data
    elif not dpkg_package_exists('ow-shell'):
  File "/home/foodie/Mycodo/mycodo/utils/system_pi.py", line 230, in dpkg_package_exists
    _, _, stat = cmd_output(cmd)
  File "/home/foodie/Mycodo/mycodo/utils/system_pi.py", line 396, in cmd_output
    pw_record = pwd.getpwnam(user)
KeyError: "getpwnam(): name not found: 'pi'"

TravisDuck999, 2021-01-16:
Hi, sorry seems to be a pretty easy fix, I just editd the file under :
/home/foodie/Mycodo/mycodo/utils/system_pi.py

I changed the user from the original "Pi" user, it seem the system is not happy with that so once I updated this file it fixed it.

In line 396 I changed both the variables in "cmd_ouput" to the correct user and it works fine now.

kizniche, 2021-01-16:
Mycodo wasn't designed to be used on operating systems other than Raspberry Pi OS. That being said, the user "pi" was removed from Mycodo several weeks ago to improve compatibility with other OSs. This change will be available in the next release. See the changelog for changes to the yet-to-be released v8.9.0.

closed 2021-02-02 by kizniche