kizniche/Mycodo #924: I298 motor controller OUTPUT for Peristaltic Pumps

Fumetin, 2021-01-21:
**Is your feature request related to a problem? Please describe.**
Hello there, im Luke, 28 yrs old chilean guy. saw ur video(Automated hydroponics) and got my brain blowed, bought all the mats that a 6yrs physician student could get for low price xD. (Raspberry 3b+ and all the sensors but most of them dont use I2C)
so trying to conect 2X "Grothe, G328, peristaltic pump 12v" whit a "I298 H motor controller" in order to control  PH

**Describe the solution you'd like**
Any chance to add an I298 output to control the peristaltic pumps? i have none experience at programing and been last 3 days learning python, so eventually ill get there but slowly.
- also I298 works whit stepper motors so wondering if the code used for DRV8825 could be used or something like that D:

**Describe alternatives you've considered**
- Save some money and get the Atlas I2c pumps for 110$ usd each  (shipment to chile is expensive haha)

- Tryed to buy a DRV 8825 and stepper motors, but in chile a stepper motor is around 100$ usd and shipment from amazon/ebay is about the same.

Mats that im using:
-Grothe, G328, peristaltic pump 12v = 22$ usd https://www.amazon.com/-/es/accesorios-dosificaci%C3%B3n-perist%C3%A1ltica-laboratorio-anal%C3%ADtica/dp/B07K1VKVBW?language=en_US
-I298N H motor controller = 5 $ usd https://www.amazon.com/-/es/control-dispositivo-escalador-corriente-Arduino/dp/B014KMHSW6/ref=sr_1_6?dchild=1&ie=UTF8&keywords=l298%20motor%20controller&language=en_US&qid=1611197621&sr=8-6

In Advance thanks for such an incredible program. :P

Fumetin, 2021-01-22:
An option could be just get a 2 channel solid state relay and conect the pumps whit the generic pump output, if im right that should work

kizniche, 2021-01-22:
The motor controller looks nice, but I think the 5-volt logic prevents it from being used with a Raspberry Pi, which uses 3.3-volt logic. It might be necessary to use a logic converter. Have you found any guides that demonstrate how to run two DC motors with a Pi?

Alternatively, as you mentioned, you can use 2 DC relays to power the pumps, in conjunction with the Generic Peristaltic Pump Output, which allows you to power the pumps for a duration and, once you determine and set the flow rate, also send volumes to dispense.

Fumetin, 2021-01-23:
L298n also works whit a Raspberry pi as it has 2 Volt logic pins, 3.3v (ENA bottom pin) and 5v (Enable A/ENA top pin), so just need to be carefully to dont conect it to the top pins xD
guide and the video:
- https://www.electronicshub.org/raspberry-pi-l298n-interface-tutorial-control-dc-motor-l298n-raspberry-pi/
- https://www.youtube.com/watch?v=2bganVdLg5Q&ab_channel=MakerTutor
- https://youtu.be/6LgFIlpOSJg?t=231   -->03:50 min. The yellow arrows shows the top pins ENA/ENB where u should NOT connect as those are 5v logic. 

Just got the 2 DC relay, ill be setting everything tonight so maybe tomorrow ill have the pumps running :P

kizniche, 2021-01-23:
Save the attached file, rename the txt extension to py, then import it from the Configure/Custom Outputs Page. You can then add this new Output from the Setup/Output Page. This output uses the code from the page you linked above. Let me know if it works, and if not, any output in the damon log that may indicate the issue.

[l298n_dc_motor.txt](https://github.com/kizniche/Mycodo/files/5859677/l298n_dc_motor.txt)

Fumetin, 2021-01-24:
wow dude u are the boss!, just tryed whit 1 pump(L298N) and worked perfectly, its 3 am and been setting everything last 14 hrs lol xD, ill get the ph sensor running tomorrow and check if its possible to regulate the ph :P.

Thanks again

Fumetin, 2021-01-25:
hey im back, just tested the PH probe/Pumps/L298N system and i just found an small issue xD, i managed to set the Controller conditional Function and the code that i bypassed since im not using EC Probe at the moment. The program didnt found any errors in the code but the pump after getting active WONT stop (the pump picked to either ph Up/Down is correct just wont stop after 1 second)

-How i Bypassed the code u gave for setting the Ec/Ph Regulation to just regulate PH: 

# Desired range for pH
range_ph_high = 7.1
range_ph_low = 6.7
# pH range that will immediately cause a pH correction
range_ph_high_danger = 7.8
range_ph_low_danger = 6.2
condition_id_measurement_ph_id = "{729860e4}"  # Condition: measurement, last, pH Input
action_id_pump_1_acid = "{062694aa}"  # Action: Pump 1 (Acid)
action_id_pump_2_base = "{cecbb852}"  # Action: Pump 2 (Base)
action_id_email_notification = "{a6df2000}"  # Action: Email Notification

### DO NOT EDIT BELOW THIS LINE UNLESS YOU KNOW WHAT YOU ARE DOING ###
import time
if 'notify_ph' not in self.variables:  # Initiate pH notification timer
    self.variables['notify_ph'] = 0
if 'notify_none' not in self.variables:  # Initiate None measurement notification timer
    self.variables['notify_none'] = 0
measure_ph = self.condition(condition_id_measurement_ph_id)

# First check if pH is dangerously low or high, and adjust if it is
if measure_ph < range_ph_low_danger:  # pH dangerously low, add base (pH up)
    msg = "pH is dangerously low: {}. Should be > {}. Dispensing 1 ml base".format(measure_ph, range_ph_low_danger)
    self.logger.debug(msg)
    self.message += msg
    self.run_action(action_id_pump_2_base)  # Dispense 1 ml base (pH up)
    if self.variables['notify_ph'] < time.time():  # Only notify every 12 hours
        self.variables['notify_ph'] = time.time() + 43200  # 12 hours
        self.run_action(action_id_email_notification, message=self.message)  # Email alert
elif measure_ph > range_ph_high_danger:  # pH dangerously high, add acid (pH down)
    msg = "pH is dangerously high: {}. Should be < {}. Dispensing 1 ml acid".format(measure_ph, range_ph_high_danger)
    self.logger.debug(msg)
    self.message += msg
    self.run_action(action_id_pump_1_acid)  # Dispense 1 ml acid (pH down)
    if self.variables['notify_ph'] < time.time():  # Only notify every 12 hours
        self.variables['notify_ph'] = time.time() + 43200  # 12 hours
        self.run_action(action_id_email_notification, message=self.message)  # Email alert

# If pH isn't dangerously low or high, make sure pH is within range
elif measure_ph < range_ph_low:  # pH too low, add base (pH up)
    self.logger.debug("pH is {}. Should be > {}. Dispensing 1 ml base".format(measure_ph, range_ph_low))
    self.run_action(action_id_pump_2_base)  # Dispense 1 ml base (pH up)
elif measure_ph > range_ph_high:  # pH too high, add acid (pH down)
    self.logger.debug("pH is {}. Should be < {}. Dispensing 1 ml acid".format(measure_ph, range_ph_high))
    self.run_action(action_id_pump_1_acid)  # Dispense 1 ml acid (pH down)

kizniche, 2021-01-25:
I just pushed an update to fix the output not turning off when turned on for a duration. See https://github.com/kizniche/Mycodo/blob/22140265595c33776e45e2e500b82a8a633afb65/mycodo/outputs/dc_motor_l298n.py. I also added the ability to set the flow rate and instruct th eoutput to pump a volume (if the motor is indeed a pump).

kizniche, 2021-01-25:
Another update. See the latest commit reference, above.

Fumetin, 2021-01-26:
Nice! Btw kinda noob here, how can i get this new output code? theres a way to update it or should i copy to a txt file and then rename it py?

kizniche, 2021-01-26:
This is the order of how I would do it:

1. Delete all the L298N outputs on the Setup->Output page.
2. Delete the output from the Configuration->Custom Output page.
3. Download and save [dc_motor_l298n.py](https://raw.githubusercontent.com/kizniche/Mycodo/5f1595fb31d000f351cc4fc61c1a99f965d3fa7b/mycodo/outputs/dc_motor_l298n.py)
4. Import the module on the Configuration->Custom Output page.
5. Add the output on the Setup->Output page.

Fumetin, 2021-01-26:
yeaah that fixed it, thank you so much :DDDDDD

kizniche, 2021-01-26:
Glad to hear. Have you tested both outputting durations and volumes?

Fumetin, 2021-01-26:
just did, both work perfectly