kizniche/Mycodo #925: PID Auto tune Function disappeared after function reorder

electricspaceman, 2021-01-25:
I was fiddly about and tried reordering the functions. I had PID autotune running but then once I clicked confirm on the reorder the function disappeared. It's still running though.

The PID autotune in the screenshot is the second one I added. If I try to activate it returns a time out. It was after this error that I realised that the original autotune is still running in the background just now showing up.

How can I disable and delete the hidden function? My heater is sort of stuck on in this state.

[image]

2021-01-25 18:57:29,279 - INFO - mycodo.daemon - Mycodo daemon v8.8.8 starting
2021-01-25 18:57:29,284 - DEBUG - mycodo.daemon - Refreshing camera settings
2021-01-25 18:57:29,365 - DEBUG - mycodo.daemon - Refreshing misc settings
2021-01-25 18:57:29,418 - DEBUG - mycodo.daemon - Anonymous statistics enabled
2021-01-25 18:57:29,465 - INFO - mycodo.pyro_daemon - Starting Pyro5 daemon
2021-01-25 18:57:29,468 - INFO - mycodo.pyro_monitor - Starting Pyro5 daemon monitor (30 min timer)
2021-01-25 18:57:29,916 - DEBUG - mycodo.daemon - Starting Output Controller
2021-01-25 18:57:30,025 - DEBUG - mycodo.controllers.controller_output - Initializing Outputs
2021-01-25 18:57:30,502 - DEBUG - mycodo.pyro_monitor - Pyro5 daemon monitor: daemon_status() response: 'alive'
2021-01-25 18:57:30,640 - INFO - mycodo.outputs.on_off_gpio_a7cb02ed - Output setup on pin 20 and turned OFF (ON=HIGH)
2021-01-25 18:57:30,641 - INFO - mycodo.outputs.on_off_gpio_a7cb02ed - Initialized in 166.1 ms
2021-01-25 18:57:30,641 - DEBUG - mycodo.controllers.controller_output - a7cb02ed (Heater) Initialized
2021-01-25 18:57:30,783 - INFO - mycodo.outputs.on_off_gpio_6d56264e - Output setup on pin 21 and turned OFF (ON=HIGH)
2021-01-25 18:57:30,784 - INFO - mycodo.outputs.on_off_gpio_6d56264e - Initialized in 140.9 ms
2021-01-25 18:57:30,784 - DEBUG - mycodo.controllers.controller_output - 6d56264e (Humidifer) Initialized
2021-01-25 18:57:30,936 - INFO - mycodo.outputs.on_off_gpio_27d0d629 - Output setup on pin 26 and turned OFF (ON=HIGH)
2021-01-25 18:57:30,936 - INFO - mycodo.outputs.on_off_gpio_27d0d629 - Initialized in 150.3 ms
2021-01-25 18:57:30,936 - DEBUG - mycodo.controllers.controller_output - 27d0d629 (Unused) Initialized
2021-01-25 18:57:30,937 - DEBUG - mycodo.controllers.controller_output - Outputs Initialized
2021-01-25 18:57:30,937 - INFO - mycodo.controllers.controller_output - Activated in 1019.2 ms
2021-01-25 18:57:31,438 - DEBUG - mycodo.daemon - Output Controller fully started
2021-01-25 18:57:31,438 - DEBUG - mycodo.daemon - Starting all activated Conditional controllers
2021-01-25 18:57:31,439 - INFO - mycodo.daemon - All activated Conditional controllers started
2021-01-25 18:57:31,439 - DEBUG - mycodo.daemon - Starting all activated Trigger controllers
2021-01-25 18:57:31,439 - INFO - mycodo.daemon - All activated Trigger controllers started
2021-01-25 18:57:31,439 - DEBUG - mycodo.daemon - Starting all activated Input controllers
2021-01-25 18:57:32,567 - INFO - mycodo.controllers.controller_input_702d1a43 - Activated in 749.2 ms
2021-01-25 18:57:44,361 - INFO - mycodo.controllers.controller_input_29d4d8ff - Activated in 583.4 ms
2021-01-25 18:57:45,504 - INFO - mycodo.controllers.controller_input_ecb32d5d - Activated in 589.3 ms
2021-01-25 18:57:46,568 - INFO - mycodo.controllers.controller_input_cb7fb265 - Activated in 501.5 ms
2021-01-25 18:57:46,569 - DEBUG - mycodo.inputs.ds18b20_owshell_cb7fb265 - Command: 'owread //temperature12; echo'
2021-01-25 18:57:46,570 - INFO - mycodo.daemon - All activated Input controllers started
2021-01-25 18:57:46,579 - DEBUG - mycodo.daemon - Starting all activated Math controllers
2021-01-25 18:57:46,619 - DEBUG - mycodo.inputs.ds18b20_owshell_cb7fb265 - Output: 'b'\n''
2021-01-25 18:57:46,620 - ERROR - mycodo.inputs.ds18b20_owshell_cb7fb265 - Obtaining measurement
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/inputs/ds18b20_owshell.py", line 101, in get_measurement
    temperature = float(owread_output.decode("latin1"))
ValueError: could not convert string to float: 
2021-01-25 18:57:46,635 - DEBUG - mycodo.inputs.ds18b20_owshell_cb7fb265 - Command: 'owread //temperature12; echo'
2021-01-25 18:57:46,674 - DEBUG - mycodo.inputs.ds18b20_owshell_cb7fb265 - Output: 'b'\n''
2021-01-25 18:57:46,676 - ERROR - mycodo.inputs.ds18b20_owshell_cb7fb265 - Obtaining measurement
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/inputs/ds18b20_owshell.py", line 101, in get_measurement
    temperature = float(owread_output.decode("latin1"))
ValueError: could not convert string to float: 
2021-01-25 18:57:46,768 - DEBUG - mycodo.controllers.controller_input_cb7fb265 - Adding measurements to InfluxDB with ID cb7fb265-4393-4e2a-bb5b-c0e83a9cb66e: {}
2021-01-25 18:57:47,169 - INFO - mycodo.controllers.controller_math_626b5184 - Activated in 95.0 ms
2021-01-25 18:57:47,170 - INFO - mycodo.daemon - All activated Math controllers started
2021-01-25 18:57:47,170 - DEBUG - mycodo.daemon - Starting all activated PID controllers
2021-01-25 18:57:47,170 - INFO - mycodo.daemon - All activated PID controllers started
2021-01-25 18:57:47,170 - DEBUG - mycodo.daemon - Starting all activated LCD controllers
2021-01-25 18:57:47,171 - INFO - mycodo.daemon - All activated LCD controllers started
2021-01-25 18:57:47,171 - DEBUG - mycodo.daemon - Starting all activated Custom controllers
2021-01-25 18:57:47,657 - INFO - mycodo.utils.PID_hirschmann.pid_autotune - PID Autotune started
2021-01-25 18:57:47,658 - INFO - mycodo.functions.function_pid_autotune_527f04d6 - Activated in 166.7 ms
2021-01-25 18:57:47,659 - INFO - mycodo.functions.function_pid_autotune_527f04d6 - PID Autotune started with options: Measurement Device: 29d4d8ff-4a58-4d0c-98ef-9cb74cbcb881, Measurement: f8ccef95-b5f1-4861-9b9b-59e679f6bfad, Output: a7cb02ed-b6f3-48c7-93b9-0f71f614f7bb, Output_Channel: 0, Setpoint: 16.0, Period: 30, Noise Band: 0.5, Outstep: 120.0, DIrection: raise
2021-01-25 18:57:47,660 - INFO - mycodo.daemon - All activated Custom controllers started
2021-01-25 18:57:47,671 - DEBUG - mycodo.daemon - Starting Widget Controller
2021-01-25 18:57:47,828 - INFO - mycodo.utils.PID_hirschmann.pid_autotune - 
2021-01-25 18:57:47,829 - INFO - mycodo.utils.PID_hirschmann.pid_autotune - Cycle: 1
2021-01-25 18:57:47,832 - INFO - mycodo.utils.PID_hirschmann.pid_autotune - switched state: relay step down
[... 9 lines cut ...]
2021-01-25 18:57:49,713 - INFO - mycodo.daemon - Mycodo daemon started in 9.780 seconds
2021-01-25 18:57:49,722 - INFO - mycodo.daemon - 49.00 MB RAM in use
2021-01-25 18:58:01,471 - DEBUG - mycodo.inputs.ds18b20_owshell_cb7fb265 - Command: 'owread //temperature12; echo'
2021-01-25 18:58:01,545 - DEBUG - mycodo.inputs.ds18b20_owshell_cb7fb265 - Output: 'b'\n''
2021-01-25 18:58:01,547 - ERROR - mycodo.inputs.ds18b20_owshell_cb7fb265 - Obtaining measurement
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/inputs/ds18b20_owshell.py", line 101, in get_measurement
    temperature = float(owread_output.decode("latin1"))
ValueError: could not convert string to float: 
2021-01-25 18:58:01,548 - DEBUG - mycodo.inputs.ds18b20_owshell_cb7fb265 - Command: 'owread //temperature12; echo'
2021-01-25 18:58:01,582 - DEBUG - mycodo.inputs.ds18b20_owshell_cb7fb265 - Output: 'b'\n''
2021-01-25 18:58:01,583 - ERROR - mycodo.inputs.ds18b20_owshell_cb7fb265 - Obtaining measurement
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/inputs/ds18b20_owshell.py", line 101, in get_measurement
    temperature = float(owread_output.decode("latin1"))
ValueError: could not convert string to float: 
2021-01-25 18:58:01,603 - DEBUG - mycodo.controllers.controller_input_cb7fb265 - Adding measurements to InfluxDB with ID cb7fb265-4393-4e2a-bb5b-c0e83a9cb66e: {}
2021-01-25 18:58:11,559 - INFO - mycodo.utils.PID_hirschmann.pid_autotune - PID Autotune started
2021-01-25 18:58:11,561 - ERROR - mycodo.functions.function_pid_autotune_c5f95d8e - Cannot start PID Autotune: Could not find output channel.
2021-01-25 18:58:11,561 - INFO - mycodo.functions.function_pid_autotune_c5f95d8e - Deactivating Autotune Function
2021-01-25 18:58:11,619 - ERROR - mycodo.functions.function_pid_autotune_c5f95d8e - Deactivated unexpectedly
2021-01-25 18:58:12,006 - DEBUG - mycodo.daemon - Could not deactivate Custom controller with ID c5f95d8e-8c19-4887-baad-b548ee152de2, it's not active.
2021-01-25 18:58:16,434 - DEBUG - mycodo.inputs.ds18b20_owshell_cb7fb265 - Command: 'owread //temperature12; echo'
2021-01-25 18:58:16,467 - DEBUG - mycodo.inputs.ds18b20_owshell_cb7fb265 - Output: 'b'\n''
2021-01-25 18:58:16,468 - ERROR - mycodo.inputs.ds18b20_owshell_cb7fb265 - Obtaining measurement

kizniche, 2021-01-25:
I just pushed a fix. I also found a hidden autotune controller once I made the fix to my dev system. This affected all custom functions, btw. Additionally, you can control the daemon with the mycodo-client command. See the help with "mycodo-client --help" for how to deactivate controllers. You can either upgrade to master to get the latest code, or use the Mycodo client.

electricspaceman, 2021-01-26:
Thanks for the fix, the hidden function has ceased (presumably it's not there anymore). I tried activating the second PID autotune (mentioned above) and it still returned a timeout.  Is something wrong with the autotune? My heating system has a really high thermal intertia so working to get the perfect PID values is going to be tricky (and time lengthy). Any advise would be appreciated as bang bang is going to only get me so far (probably oscillate quite a lot). I've read everything you've written about the values and what they do (several times lol) but I still find it a little intimidating. 

My system is no longer stuck 'on' though which is great! Thanks for the help Kyle.

closed 2021-03-09 by kizniche