kizniche/Mycodo #926: DS18S20 - not showing any temp values

jbrooks4, 2021-01-29:
I have three sensors hooked up to the 1wire pin on a RPI4. The IDs show up in the list, but on the live data page the sensors just show 0. I tested via the command line as well, and get results for each. I tested an input configuration for the seeed library that I found on the forums and that seems to show data on one the sensors. The logs for Mycodo show the following for the regular DS18B20 input though. 
There was a similar issue that was closed. #796. Do I just need to upgrade to the master ? 

(test from the Pi) 
pi@raspberrypi:/sys/bus/w1/devices/28-0417716357ff $ cat w1_slave
3d 01 4b 46 7f ff 0c 10 75 : crc=75 YES
3d 01 4b 46 7f ff 0c 10 75 t=19812

2021-01-29 08:35:00,104 - INFO - mycodo.daemon - All Custom controllers stopped
2021-01-29 08:35:00,104 - INFO - mycodo.daemon - All LCD controllers stopped
2021-01-29 08:35:00,104 - INFO - mycodo.daemon - All PID controllers stopped
2021-01-29 08:35:00,104 - INFO - mycodo.daemon - All Math controllers stopped
2021-01-29 08:35:00,105 - INFO - mycodo.daemon - All Input controllers stopped
2021-01-29 08:35:00,105 - INFO - mycodo.daemon - All Trigger controllers stopped
2021-01-29 08:35:00,105 - INFO - mycodo.daemon - All Conditional controllers stopped
2021-01-29 08:35:00,121 - INFO - mycodo.controllers.controller_output - Deactivated in 15.6 ms
2021-01-29 08:35:00,208 - INFO - mycodo.controllers.controller_widget - Deactivated in 86.7 ms
2021-01-29 08:35:00,208 - INFO - mycodo.daemon - Mycodo daemon terminated in 1.097 seconds

2021-01-29 08:35:19,459 - INFO - mycodo.daemon - Mycodo daemon v8.8.8 starting
2021-01-29 08:35:19,804 - INFO - mycodo.controllers.controller_output - Activated in 116.3 ms
2021-01-29 08:35:20,305 - INFO - mycodo.daemon - All activated Conditional controllers started
2021-01-29 08:35:20,305 - INFO - mycodo.daemon - All activated Trigger controllers started
2021-01-29 08:35:20,306 - INFO - mycodo.daemon - All activated Input controllers started
2021-01-29 08:35:20,306 - INFO - mycodo.daemon - All activated Math controllers started
2021-01-29 08:35:20,306 - INFO - mycodo.daemon - All activated PID controllers started
2021-01-29 08:35:20,306 - INFO - mycodo.daemon - All activated LCD controllers started
2021-01-29 08:35:20,306 - INFO - mycodo.daemon - All activated Custom controllers started
2021-01-29 08:35:20,642 - INFO - mycodo.controllers.controller_widget - Activated in 335.1 ms
2021-01-29 08:35:21,643 - INFO - mycodo.daemon - Mycodo daemon started in 2.181 seconds
2021-01-29 08:35:21,649 - INFO - mycodo.daemon - 39.06 MB RAM in use
2021-01-29 08:36:27,010 - ERROR - mycodo.inputs.ds18b20_d238b954 - Input initialization
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/inputs/ds18b20.py", line 75, in initialize_input
    W1ThermSensor.THERM_SENSOR_DS18B20, self.input_dev.location)
AttributeError: type object 'W1ThermSensor' has no attribute 'THERM_SENSOR_DS18B20'
2021-01-29 08:36:27,011 - INFO - mycodo.controllers.controller_input_d238b954 - Activated in 155.5 ms
2021-01-29 08:36:27,012 - ERROR - mycodo.inputs.ds18b20_d238b954 - Input not set up
2021-01-29 08:36:42,034 - ERROR - mycodo.inputs.ds18b20_d238b954 - Input not set up
2021-01-29 08:36:56,955 - ERROR - mycodo.inputs.ds18b20_d238b954 - Input not set up
2021-01-29 08:36:56,956 - ERROR - mycodo.controllers.controller_input_d238b954 - StopIteration raised 3 times.

kizniche, 2021-01-29:
They refactored how the class operates 4 days ago and broke current implementations. 
https://github.com/timofurrer/w1thermsensor/releases/tag/v2.0.0

jbrooks4, 2021-01-29:
Interesting, any easy way to switch it over to the seeed library for all three sensors ?

kizniche, 2021-01-29:
That would be the most difficult solution. The easiest solution is to merely downgrade the library to the previous version.

jbrooks4, 2021-01-29:
I will give that a go.

kizniche, 2021-01-29:
~/Mycodo/env/bin/pip install w1thermsensor==1.0.5
Then reboot

kizniche, 2021-01-29:
I just committed fixes for all Inputs that use the w1thermsensor library. If you're willing to test on the master branch, which should be stable at this point, please [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) and report back if the Input works. Thanks.

jbrooks4, 2021-01-29:
Cheers will do, just finishing up a few things on the hardware right now.

kizniche, 2021-01-29:
FYI, the fix I implemented is for the use of the latest 2.0.0 version of w1thermsensor, not merely a downgrade to an old version.

jbrooks4, 2021-01-29:
That worked! Thanks for your help mate. 

Cheers

frodus17, 2021-02-17:
Had the same issue. Updated to master. 

However, the output and PID stopped working. I had to start over and re-add my temp sensor, GPIO output and PID.

I have to do this same thing just about every time I upgrade to the latest.

closed 2021-03-09 by kizniche