kizniche/Mycodo #927: Using MQTT Output disconnects unrelated MQTT Input (unrecoverable)

kizniche, 2021-02-03:
MQTT Inputs work as expected until an MQTT on/off Output is turned on of off. The Input is unable to operate properly after this occurs.

```
2021-01-31 19:58:08,999 - DEBUG - mycodo.inputs.mqtt_paho_751d0e5a - inp_r222 Adding measurement to influxdb: {0: {'measurement': 'volume', 'unit': 'l', 'value': 2.0, 'timestamp_utc': datetime.datetime(2021, 1, 31, 17, 58, 8, 996874)}}
2021-01-31 19:58:09,017 - DEBUG - mycodo.inputs.mqtt_paho_751d0e5a - inp_r174 Message received: Channel: esp32/temperature, Value: 24.48
2021-01-31 19:58:09,020 - DEBUG - mycodo.inputs.mqtt_paho_751d0e5a - inp_r222 Adding measurement to influxdb: {1: {'measurement': 'temperature', 'unit': 'C', 'value': 24.48, 'timestamp_utc': datetime.datetime(2021, 1, 31, 17, 58, 9, 17463)}}
2021-01-31 19:58:09,025 - DEBUG - mycodo.inputs.mqtt_paho_751d0e5a - inp_r174 Message received: Channel: esp32/humidity, Value: 26.59
2021-01-31 19:58:09,029 - DEBUG - mycodo.inputs.mqtt_paho_751d0e5a - inp_r222 Adding measurement to influxdb: {2: {'measurement': 'humidity', 'unit': 'percent', 'value': 26.59, 'timestamp_utc': datetime.datetime(2021, 1, 31, 17, 58, 9, 24994)}}
2021-01-31 19:58:09,245 - DEBUG - mycodo.outputs.mqtt_3b6f5df5 - output_on_off(on, 0, sec, 3.0, 0.0, True)
2021-01-31 19:58:09,247 - DEBUG - mycodo.inputs.mqtt_paho_751d0e5a - inp_r229 input Disconnected result code 1
2021-01-31 19:58:09,249 - DEBUG - mycodo.outputs.mqtt_3b6f5df5 - Output 3b6f5df5-19c9-4bde-a312-34e2370ff978 CH0 (esp32/output) on for 3.0 seconds. Output returned: None
2021-01-31 19:58:10,249 - DEBUG - mycodo.inputs.mqtt_paho_751d0e5a - r162 Connected to 'None', rc: 0
2021-01-31 19:58:12,250 - DEBUG - mycodo.outputs.mqtt_3b6f5df5 - output_on_off(off, 0, None, 0.0, 0.0, True)
2021-01-31 19:58:12,254 - DEBUG - mycodo.inputs.mqtt_paho_751d0e5a - inp_r229 input Disconnected result code 1
2021-01-31 19:58:12,256 - DEBUG - mycodo.outputs.mqtt_3b6f5df5 - Output 3b6f5df5-19c9-4bde-a312-34e2370ff978 CH0 (esp32/output) OFF at 2021-01-31 19:58:12. Output returned: None
2021-01-31 19:58:13,255 - DEBUG - mycodo.inputs.mqtt_paho_751d0e5a - r162 Connected to 'None', rc: 0
```

kizniche, 2021-02-09:
Issue was input and output using the same Client ID

closed 2021-02-09 by kizniche

garudaonekh, 2021-07-14:
@kizniche so client_id should be given unique client id in case user forgets to change its name.