kizniche/Mycodo #929: possible wiringpi / pigpio dependency

ebo, 2021-02-04:
While working to get Mycodo working on Gentoo I came across a note that the maintainer of wiringpi has depreciated the project, and pigpio is listed as the replacement.  Searching around Mycodo source code it looks like both are installed and still used.  Is this correct, or is wiringpi deprecated and there are still a few bits that need replacing/removed?

kizniche, 2021-02-09:
It looks like only one input (ADS1256) uses it. It's only installed during setup for the status page where it displays the output of ```gpio readall```

ebo, 2021-02-09:
I do not know enough of the newer pigpio dependency.  Looking at some of the online posts, it looks like pigpio supports the ADS1256 (for starters):

  https://www.raspberrypi.org/forums/viewtopic.php?t=134270&start=25
  https://www.mikrocontroller.net/attachment/356051/ADS1256.pdf

Do you know someone that has a ADS1256 that could test a pigpio replacement of the "gpio readall" functionality?  I found the following link that reports to have developed an equivalent:

  https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=254071

Since I do not have the board, I cannot test the replacement.

closed 2021-09-20 by kizniche

ebo, 2021-09-20:
Gack... This project got sidelined with so many pandemic distractions.  It looks like I really need to pull this off the back-40's backburner.  Thanks for poking at this everyone that has.