kizniche/Mycodo #931: Setup Failed Around Webpage Frontend

linguine2552, 2021-02-08:
```
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 5 times)...
#### Trying again...

#### Setting permissions

#### Restarting the Mycodo daemon
Traceback (most recent call last):
  File "/home/dietpi/Mycodo/mycodo/scripts/restart_daemon.py", line 12, in <module>
    from mycodo.config import DAEMON_PID_FILE
  File "/home/dietpi/Mycodo/mycodo/config.py", line 10, in <module>
    from flask_babel import lazy_gettext
ModuleNotFoundError: No module named 'flask_babel'
```
Might be because I'm using a pi0 over wifi.

kizniche, 2021-02-08:
> ModuleNotFoundError: No module named 'flask_babel'

For some reason your Python environment didn't install flask-babel properly. Try installing the requirements from a terminal with ```~/Mycodo/env/bin/pip install -r ~/Mycodo/install/requirements.txt```

linguine2552, 2021-02-08:
installing the requirements didn't work however I was successful in installing Mycodo on a different pi0 with the standard raspbianOS. The pi0 I used that kept failing was flashed with dietPi, probably kept failing because it was missing some core programs.

closed 2021-02-09 by kizniche