kizniche/Mycodo #932: MQTT Input Action Publish

andrewoke, 2021-02-15:
**Is your feature request related to a problem? Please describe.**
I would like to be able to publish all my sensor readings via mqtt. At this point I see I can use the mqtt value output, but it requires I set a specific value. Using a function I can't seem to grab the value and publish it.

**Describe the solution you'd like**
I would like to be able to add an mqtt output and in the input configure (currently miflora sensors) link them together.

**Describe alternatives you've considered**
Have read your forum posts and github issues. There is talk of input actions so I thought I would create a specific issue so I don't need to remember where to look.

**Additional context**

kizniche, 2021-02-15:
We've discussed Inputs being able to utilize MQTT in the past. My solution was to add the ability to add Actions to Inputs. However, I have yet to add this functionality.

andrewoke, 2021-02-15:
Yeah, I've seen that. A couple points. a) I'm happy to sponsor work on this with a (larger) donation and b) Is there a workaround to do the above?

kizniche, 2021-02-16:
The quickest way to get this functionality would be to copy the Input module from Mycodo/mycodo/inputs and modify the unique name and add the MQTT Python code, then import the new module from the Configure -> Custom Inputs page.

I do see a lot of utility in allowing MQTT for any input, but I've been grappling with the best way to implement it. I'll chew on it tonight and get back to you with what I come up with.

kizniche, 2021-04-07:
This issue has been mentioned on **KG**. There might be relevant details there:

https://forum.kylegabriel.com/t/sending-a-value-to-the-mqtt-hub/57/2

kizniche, 2021-09-08:
This issue has been mentioned on **DIY Automation Forum**. There might be relevant details there:

https://forum.kylegabriel.com/t/sending-or-requesting-measurements-from-one-mycodo-instance-to-another/358/2

kizniche, 2022-03-23:
I recently added the ability to add Actions to Inputs, with the first Input Action being MQTT Publish. You can [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) if you want to try the code prior to the 8.13.0 release.