kizniche/Mycodo #933: PCA9685 Output Module does not work (Version 0.8.9)

pkrahmer, 2021-02-15:
### Describe the problem/bug

pca9685 output module issues:
- Adding device fails, because it tries to reference "custom_options" in OUTPUT_INFORMATION but it is not present. Instead, two sections called "custom_channel_options" are present.
- As to the datasheet, "All outputs are set to the same PWM frequency." The implementation is ambiguous, as it reads a potenially different value per channel but writes all to the same method.
- No name per channel can be given (e.g. this works for pcf8574 output module)
- On pca9685 initialization, my device defaults all outputs to 0. Therefore state_startup "do nothing" does not work and should be removed, default should be set to "Off", as this is the devices default behaviour.
- inverted output also inverts reported value and mixes up on-off state handling. inversion should only apply just before setting value to device driver, not for the reported values
- value is set using "self.pwm_output.set_pwm(self.options_channels['pwm_hertz'][output_channel], amount)". As to the adafruit documentation, set_pwm has three parameters: 
-- channel: The channel that should be updated with the new values (0..15)
-- on: The tick (between 0..4095) when the signal should transition from low to high
-- off:the tick (between 0..4095) when the signal should transition from high to low
- correct implementation could be:
-- off_at_tick = round(output_amount*4095./100.)
-- self.pwm_output.set_pwm(output_channel, 0, off_at_tick)

### Versions:

 - Mycodo Version: 0.8.9 (commit dcf4c102eebb24e5569ffc1d5a98b8c472681968)
 - Raspberry Pi Version: 3B+
 - Raspbian OS Version: Buster Lite

### Additional context

I will submit a pull request with a change suggestion.

pkrahmer, 2021-02-15:
Please see suggested bugfix at https://github.com/kizniche/Mycodo/pull/934

pkrahmer, 2021-02-15:
Hi Kyle. Have been in a hurry when submitting the pull request and issue report. Just had another look on it and noticed it sounds a bit rude. :-) 
First, thanks for you marvelous and for sure intensive work creating mycodo.
I'm quite new using it. Originally started to develop an own control software for a small seedling box. I found mycodo and am now experimenting if it supports all my hardware. 
I'm a software professional (but not a python pro) so I tried to fix the pca issue mainly to get it running at my end. 
This issue report and pull request is to give back by participation. If you have any concerns in the way I implemented the changes, please let me know. I'm happy to make the necessary changes and resubmit if you like.
Pascal

kizniche, 2021-02-15:
Nice work. Thanks for the PR. This was a module I quickly threw together and hadn't had a chance to thoroughly go over or test (I don't have this device). It looks like you fixed all the issues. Have you tested and verified it now works as expected?

pkrahmer, 2021-02-15:
Yes, I've got it running here for my 7 (out of the 16) used output channels. 5 LED dimmer, 2 fans. Working good.