kizniche/Mycodo #937: trigger_run_pwm_method does not work (yet-to-be-released v0.8.9)

pkrahmer, 2021-02-16:
### Describe the problem/bug

Using trigger with run_pwm_method fails for several reasons. Please see PR review comments for details.
https://github.com/kizniche/Mycodo/pull/936

Changes tested with following test scenarios:
- Method without repeat
- Method with infinite repeat
- Method with time limited repeat
- Bounce backend during running controller
- Reboot system during running controller

### Versions:

 - Mycodo Version: 0.8.9 (commit dcf4c102eebb24e5569ffc1d5a98b8c472681968)
 - Raspberry Pi Version: 3B+
 - Raspbian OS Version: Buster Lite

### Additional context

I will submit a pull request with a change suggestion.

kizniche, 2021-02-16:
Nice work. Thanks for catching all those issues. I'm actually surprised that this used to work... I know I tested it to work at one point in time. Everything looks good except I think your change from self.method_start_act to self.method_start_time isn't necessary.

pkrahmer, 2021-02-16:
You're welcome. Its actually much less work than me reinventing the wheel, so it's a good deal :-).
Based on the commit history, it seems this came in when "multiple output channels" were added.

self.method_start_act to self.method_start_time was my first solution but then I found the root cause and missed to revert this. But it shouldn't make a difference, as the assignment is done just a couple of lines above.

pkrahmer, 2021-02-16:
By the way, I'm investigating ways to allow the daily schedule methods to be used with run_pwm_method (to dim my lights nicely on a daily basis). Any concerns about that?

kizniche, 2021-02-16:
> I'm investigating ways to allow the daily schedule methods to be used with run_pwm_method (to dim my lights nicely on a daily basis). Any concerns about that?

I don't see an issue. You could also use two Daily Point Timers and the Output (Ramp Duty Cycle) Actions to accomplish the same, unless you wanted a curved ramp, and in that case a Method would work better.

pkrahmer, 2021-02-16:
Interesting approach, will give it a try first.

closed 2021-03-09 by kizniche