kizniche/Mycodo #940: Issue when timer fuction

os-is-awesome, 2021-02-22:
Hello, When trying to deactivate fuctions, so far only tested 3 different timer triggers.  UI shows sucess in deactivating but then shows error.  And trigger remains on while ui visually showing as off.  Backend restart was the only way i could correct it.    
If there is something you want me to try please let me know or if i am doing something wrong.  
Thanks 

 - Mycodo Version: [e.g. current on github]
 - Raspberry Pi Version: [e.g. 4b+]
 - Raspbian OS Version: [e.g. Buster]

Pl
______ - 8.8.8
02/22 07:33
Select the number of lines to display from the end of a log
Last 30 lines of /var/log/mycodo/mycodo.log:

2021-02-22 07:33:03,781 - DEBUG - mycodo.controllers.controller_output - output_on_off(1a80ef6a-78f3-4d42-8f1b-659c53aa2a33, on, 0, sec, 10.0, 0.0, True)
2021-02-22 07:33:03,782 - DEBUG - mycodo.outputs.on_off_gpio_1a80ef6a - output_on_off(on, 0, sec, 10.0, 0.0, True)
2021-02-22 07:33:03,782 - DEBUG - mycodo.outputs.on_off_gpio_1a80ef6a - Output 1a80ef6a-78f3-4d42-8f1b-659c53aa2a33 CH0 (Air solenoid) is already on for an amount of 10.00 seconds (with 3.92 seconds remaining). Recording the amount of time the output has been on (6.08 sec) and updating the amount to 10.00 seconds.
**### 2021-02-22 07:33:11,263 - ERROR - mycodo.daemon - Could not deactivate Trigger controller with ID babdb30e-f8c2-490c-9d48-6ccf9b20a291: 'NoneType' object has no attribute 'channel'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 341, in controller_deactivate
    self.controller[cont_type][cont_id].stop_controller()
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 109, in stop_controller
    self.pre_stop()
  File "/var/mycodo-root/mycodo/controllers/controller_trigger.py", line 475, in pre_stop
    output_channel=output_channel.channel)
AttributeError: 'NoneType' object has no attribute 'channel**
2021-02-22 07:33:13,827 - DEBUG - mycodo.outputs.on_off_gpio_1a80ef6a - output_on_off(off, 0, None, 0.0, 0.0, True)
2021-02-22 07:33:13,827 - DEBUG - mycodo.outputs.on_off_gpio_1a80ef6a - Output 1a80ef6a-78f3-4d42-8f1b-659c53aa2a33 CH0 (Air solenoid) OFF at 2021-02-22 07:33:13. Output returned: success
2021-02-22 07:33:16,378 - DEBUG - mycodo.inputs.dht11_715e3d6f - Exception raised when taking a reading: Bad Checksum.
2021-02-22 07:33:16,609 - DEBUG - mycodo.controllers.controller_input_715e3d6f - Adding measurements to InfluxDB with ID 715e3d6f-28ec-46db-88bc-c1b721b7cd26: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 23.0, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 49.0, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 11.707258100708337, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 1432.5011665323732, 'timestamp_utc': None}}
2021-02-22 07:33:17,675 - DEBUG - mycodo.trigger_action_8d2ba1db - Message: 2021-02-22 07:33:17

kizniche, 2021-02-24:
Thanks for the bug report. I'll make a fix for the next release.

os-is-awesome, 2021-02-24:
Your welcome! Thank you for the awesome software!

closed 2021-03-09 by kizniche