kizniche/Mycodo #941: output to Wifi Relay

chamroeunbm, 2021-02-24:
**Is your feature request related to a problem? Please describe.**
I want to output to Tasmota relay.
I found two options: 1 is via http command and another is MQTT. which one would you recommend?
Secondly, I need to make it work like momentary button so that if Mycodo restart or turn off accidentally, the relay is off, i donot the relay to be in orphan state. My initial solution for this second iasue is to send 5 second pulse and keep sending it every 4 second to keep the relay on. But i donot know how to keep repeating like this in  custom output module. Or do you have any better suggestion?

kizniche, 2021-02-25:
I've used both methods. The http method doesn't require having an MQTT server running. The second method requires an MQTT server, but this is as easy as installing mosquitto (```sudo apt install mosquitto```) on your Pi hosting Mycodo. The issue with wireless relays is you cannot guarantee they turn on when instructed (well, you also can't guarantee wired ones turn on also, but it's much less likely for communication to fail). There was talk a while ago about creating a way to check the state of a wireless output for just this reason (#728), but it hasn't really gone anywhere.

chamroeunbm, 2021-03-11:
Thanks. I think HTTP is better for this as it eliminate another point of failure if MQTT is not working and burden.

closed 2021-03-11 by chamroeunbm