kizniche/Mycodo #942: "404: Not found" when I try to install Mycodo

yvesegli, 2021-02-26:
Hi!

I'm doing my very first steps with my Raspberry Pi and I currently try to install Mycodo because Im building my own automated hydroponic system. Please apologize my beginner question, but I can't solve my problem, even though I researched a lot of things...

Thanks for your help.

Best regards from Switzerland,
Yves

### Describe the problem/bug

Not able to install Mycodo

### Versions:

--

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. curl -L https://kizniche.github.io/Mycodo/install
--> then it says, I should execute the following (if I didn't download/extract Mycodo yet, which is the case)
2. curl -L https://raw.githubusercontent.com/kizniche/Mycodo/master/install/install
--> 404: Not found

### Expected behavior

Install Mycodo

### Screenshots

--

### Additional context

--

kizniche, 2021-02-26:
I don't understand. This isn't an expected response to the install script. Can you attach a screenshot?

yvesegli, 2021-03-01:
Hey

Yes of course, I attached you two screenshots which show my steps. Maybe I can give you a clew - I tested if I can call something when I type in the following URL in the browser, it also says "404 not found". I think there has to be some content, right?

Thank you very much and have a nice day.

Yves
[image]
[image]

kizniche, 2021-03-03:
> curl -L https://kizniche.github.io/Mycodo/install

You're not entering the correct command. Review the install command listed in the [install section](https://github.com/kizniche/Mycodo#install).

closed 2021-03-03 by kizniche

yvesegli, 2021-03-03:
I do - in a first step I enter "curl -L https://kizniche.github.io/Mycodo/install" and then I get further instructions (please see my attached screenshots). As a second step I enter "curl -L https://raw.githubusercontent.com/kizniche/Mycodo/master/install/install" because it's described to enter this command if Mycodo isn't downloaded/extracted yet. What do I do wrong?

Thanks for your help.

kizniche, 2021-03-03:
You're not entering the entire command. Look at it carefully and you'll see you're only entering part of the command.