kizniche/Mycodo #943: Error while loading the web interface

Kennedys1337, 2021-03-07:
Fehler 500: interner Server-Fehler

Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 639, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 1196, in page_live
    use_unit=use_unit)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 140, in render_template
    ctx.app,
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 120, in _render
    rv = template.render(context)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 1090, in render
    self.environment.handle_exception()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 832, in handle_exception
    reraise(*rewrite_traceback_stack(source=source))
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/_compat.py", line 28, in reraise
    raise value.with_traceback(tb)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/live.html", line 3, in top-level template code
    {% set help_page = ["https://kizniche.github.io/Mycodo/Data-Viewing/#live-measurements", _('Live Measurements')] %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/layout.html", line 327, in top-level template code
    {%- block body %}{% endblock -%}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/live.html", line 167, in block "body"
    {%- if dict_units[dict_measure_units[each_meas.unique_id]]['unit'] -%}
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 452, in getitem
    return obj[argument]
jinja2.exceptions.UndefinedError: 'collections.OrderedDict object' has no attribute ''

Kennedys1337, 2021-03-07:
I just upgraded to master and used the multi measure equation, now the live page is comming up with this error.

kizniche, 2021-03-07:
The error occurs because you don't have a measurement/unit set for the function output. I still need to add checks that the user set those before allowing activation.

Kennedys1337, 2021-03-07:
Cheers

closed 2021-03-07 by Kennedys1337

kizniche, 2021-03-07:
You could have left it open... I'm in the middle of fixing it, so it doesn't matter too much now, but if I wasn't it's good to keep the issue open until resolved. In any case, thanks for the bug report and I'll have a commit pushed shortly to fix the issue.

kizniche, 2021-03-07:
fix pushed. Give it a try and let me know if there are any issues.

Kennedys1337, 2021-03-07:
Yes, it is working! 

Error: Activate Controller: Measurement CH0 (Equation Output) measurement/unit not set. All Measurements need to have the measurement and unit set before the Function can be activated.

neongrau81, 2021-03-10:
Hey @kizniche I am facing a similar issue when I try to navigate to http://ip.ad.dr.ess/input after updating to 8.9.0 

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 639, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 2135, in page_input
    devices_1wire_w1thermsensor=devices_1wire_w1thermsensor)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 140, in render_template
    ctx.app,
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 120, in _render
    rv = template.render(context)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/asyncsupport.py", line 76, in render
    return original_render(self, *args, **kwargs)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 1008, in render
    return self.environment.handle_exception(exc_info, True)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 780, in handle_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/_compat.py", line 37, in reraise
    raise value.with_traceback(tb)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/input.html", line 3, in top-level template code
    {% set help_page = ["https://kizniche.github.io/Mycodo/Inputs/", dict_translation['input']['title']] %}
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 1005, in render
    return concat(self.root_render_func(self.new_context(vars)))
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/input.html", line 22, in root
    <input type="hidden" name="form-name" value="addSensor">
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/layout.html", line 333, in root
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/input.html", line 74, in block_body
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/data_options/math.html", line 141, in root
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/data_options/math_options/TEMPLATE_OPTIONS_Measurements_Convert.html", line 149, in root
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 411, in getitem
    return obj[argument]
jinja2.exceptions.UndefinedError: 'collections.OrderedDict object' has no attribute 'jPa'

I had a controller without output and I already tried activating / deactivating but that didn't solve my issue.
Do you have any idea?

Thanks in advance