kizniche/Mycodo #946: Cannot save MQTT input configuration changes

danielfppps, 2021-03-10:
I am using Mycodo 8.9.0. Whenever I add an MQTT input, click the (+) sign and then click "Save" - no changes made at all - I receive the following error :

Error: Modify Input: '<' not supported between instances of 'NoneType' and 'float'

The error happens regardless of which fields I change or attempt to modify (even when no fields are modified as I mentioned above). I'm new to mycodo so please do let me know if I'm doing something wrong. Thanks in advance for your help!

kizniche, 2021-03-10:
I'll take a look. Thanks for the bug report.

chamroeunbm, 2021-03-11:
Same issue here.

closed 2021-03-13 by kizniche

chamroeunbm, 2021-03-16:
hi @kizniche , still bug in utils_input.py on line 484. 

it should be new_channel.channel rather than new_measurement.channel = index

kizniche, 2021-03-16:
>it should be new_channel.channel rather than new_measurement.channel = index

Good catch. just pushed a fix. Thanks.