kizniche/Mycodo #947: Error 500 wen accessing Inputs page

neongrau81, 2021-03-10:
### Describe the problem/bug

After updating from 8.8.8 to 8.9.0 I can no longer access the Inputs page. Everything else seems to work fine
 

### Versions:

 - Mycodo Version: 8.9.0
 - Raspberry Pi Version: 4 B+
 - Raspbian OS Version: Buster 

### Additional context
Error (Full Traceback):

Traceback (most recent call last):
File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
response = self.full_dispatch_request()
File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
rv = self.handle_user_exception(e)
File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 639, in error_router
return original_handler(e)
File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
reraise(exc_type, exc_value, tb)
File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
raise value
File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
rv = self.dispatch_request()
File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
return self.view_functionsrule.endpoint
File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 272, in decorated_view
return func(*args, **kwargs)
File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 2135, in page_input
devices_1wire_w1thermsensor=devices_1wire_w1thermsensor)
File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 140, in render_template
ctx.app,
File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 120, in _render
rv = template.render(context)
File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/asyncsupport.py", line 76, in render
return original_render(self, *args, **kwargs)
File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 1008, in render
return self.environment.handle_exception(exc_info, True)
File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 780, in handle_exception
reraise(exc_type, exc_value, tb)
File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/_compat.py", line 37, in reraise
raise value.with_traceback(tb)
File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/input.html", line 3, in top-level template code
{% set help_page = ["https://kizniche.github.io/Mycodo/Inputs/", dict_translation['input']['title']] %}
File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 1005, in render
return concat(self.root_render_func(self.new_context(vars)))
File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/input.html", line 22, in root

File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/layout.html", line 333, in root
File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/input.html", line 74, in block_body
File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/data_options/math.html", line 141, in root
File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/data_options/math_options/TEMPLATE_OPTIONS_Measurements_Convert.html", line 149, in root
File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 411, in getitem
return obj[argument]
jinja2.exceptions.UndefinedError: 'collections.OrderedDict object' has no attribute 'jPa'

I had a controller without output and I already tried activating / deactivating but that didn't solve my issue

kizniche, 2021-03-10:
Do you know what "jPa" is? Is this a unit you created from the Configure menu?

neongrau81, 2021-03-11:
the "jPa" looks a lot like a mis-spelled "kPa" but from what I can see the only custom conversion I use is for my water tank (where i measure the distance between sensor and water surface to calculate how much water is left).

neongrau81, 2021-03-12:
I might found something :-D 
When I was searching the files on my pi for the string "jPa" there was exactly one result in ._cargo/registry/index/github.com-1ecc6299db9ec823/.git/objects/pack/pack-8e1b3ef4d282eec507eacb95eb08ab68aba9a7f0.pack_

Then i searched here in github and found the string "jPa" in the file mycodo/config_devices_units.py
[image]

 Knowing this I checked the /measurements page there is a conversion from kPa to something else.
[image]
Looks like the something else could/should be the jPa.

kizniche, 2021-03-12:
Interesting. Does deleting that line and restarting the frontend fix the issue?

neongrau81, 2021-03-12:
I can't delete the line but I checked an old version of the file the right conversion in mycodo/config_devices_units.py seems to be ('kPa', 'hPa', 'x*10'),

kizniche, 2021-03-12:
What do you mean you can't delete the line? We need to determine if removing the offending line resolves the issue or creates another issue.

neongrau81, 2021-03-12:
Ha, okay now i got it. You were talking about deleting the line in mycodo/config_devices_units.py I was looking at the line /measurement :-)

I changed in config_devices_units.py line 540 from ('kPa', 'jPa', 'x*10') to ('kPa', 'hPa', 'x*10') 
-> after I rebooted the pi i experience the same error
removing the line and restarting the Pi -> same error

kizniche, 2021-03-12:
Okay, it looks like you'll need to create a dummy "jPa" unit with the rest of the units in that file. I'll create a database upgrade the fixes it when I get a chance to make a bug fix release, but in the meantime adding the dummy unit should fix the issue.

neongrau81, 2021-03-12:
Thanks once again @kizniche!
Adding a dummy 'jPa' unit solves the issue

kizniche, 2021-03-13:
I haven't been able to replicate this issue. Can you provide a bit more information about your install? What version did you originally install? If you upgraded and this issue began, what version did you upgrade from? What Inputs did you have installed when this issue began? I need to find the system configuration that this issue presents itself to be able to develop a solution to ensure other users don't experience this or it's fixed with a database change.

closed 2021-03-13 by kizniche