kizniche/Mycodo #949: Upgrading from 8.8.8 to latest 8.9.1 results in dictionary error

rbbrdckybk, 2021-03-14:
Just upgraded a basically fresh install of MyCodo 8.8.8 to 8.9.1 via the web interface. Upgrade completed successfully, no errors. After the upgrade, navigating to most pages results in the following HTTP 500 error (I'm new to MyCodo, but vaguely remember creating a custom measurement that I tagged "999" when I was originally looking things over - so I assume the dictionary error has something to do with that):

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 639, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 1200, in page_live
    use_unit=use_unit)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 140, in render_template
    ctx.app,
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 120, in _render
    rv = template.render(context)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 1090, in render
    self.environment.handle_exception()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 832, in handle_exception
    reraise(*rewrite_traceback_stack(source=source))
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/_compat.py", line 28, in reraise
    raise value.with_traceback(tb)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/live.html", line 3, in top-level template code
    {% set help_page = ["https://kizniche.github.io/Mycodo/Data-Viewing/#live-measurements", _('Live Measurements')] %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/layout.html", line 327, in top-level template code
    {%- block body %}{% endblock -%}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/live.html", line 118, in block "body"
    {{' (' + dict_measurements[each_meas.measurement]['name'] + ') '}}
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 452, in getitem
    return obj[argument]
jinja2.exceptions.UndefinedError: 'collections.OrderedDict object' has no attribute '999'

kizniche, 2021-03-14:
It looks like one of your inputs may be referencing a missing measurement.

rbbrdckybk, 2021-03-14:
That is probably it. I set MyCodo up as a fresh 8.8.8 install a couple weeks ago, and set up a single input to test things. I'm pretty sure I created a custom measurement, and I probably set my input to reference it. Everything has been working properly for weeks, until the upgrade to 8.9.1 yesterday. The upgrade completed successfully without errors, but I can't navigate to most pages now - I just get this error. I'm pretty sure something happened during the upgrade that dropped or otherwise broke the custom measurement.

I'll probably just re-install MyCodo since I didn't have much set up anyway - just wanted to give you a heads up!

kizniche, 2021-03-15:
I created a database upgrade script that will remove that conversion in the next release. You're probably one of only a few that may experience this issue.

kizniche, 2021-03-16:
just made a release that fixes the errors on the Input/Function pages when there are missing measurements/units and will indicate there's an issue when the options are expanded.

closed 2021-03-16 by kizniche