kizniche/Mycodo #950: Python 3 code input not storing measurements after upgrading to 8.9.1

rbbrdckybk, 2021-03-14:
### Issue Summary

A Python 3 code input that was successfully running in Mycodo 8.8.8 without issues is now not storing measurements to the database. Mycodo is logging many of these errors:

`ERROR - mycodo.controllers.controller_input_957ddbe6 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.`

The code DOES successfully return the expected value, it just doesn't get stored. I can see the returned value in the Data -> Live view, and also on any of the dashboard widgets that use the live value. I can NOT view historical data (e.g. on either of the graph types) as I could on 8.8.8.

This is on a fresh install of Mycodo 8.9.1 with a single Python 3 input configured in the same manner (with the same code) that it was running successfully on 8.8.8.

### Versions:

 - Mycodo Version: 8.9.1
 - Raspberry Pi Version: 4
 - Raspbian OS Version: Buster Lite 2020-02-13

### Reproducibility

The Python 3 code manipulates a JSN-SR04T sensor, so you'd need one to test. On a side note, I see that the latest version of Mycodo supports this sensor natively, but my environment has enough background noise the the sensor results are far too inconsistent to use without some manipulation, which I accomplish via code.

1. Go to 'Setup' -> 'Input'
2. Click on 'Linux: Python 3 Code: Store Value(s) [Mycodo]'
3. Configure as shown in this screenshot:
[image]
The full Python 3 code:
```
import RPi.GPIO as GPIO
import time

# Define container limit lengths in cm
DISTANCE_TO_FULL = 35       # maximum water level; basically the min valid range
DISTANCE_TO_EMPTY = 74.5    # minimum water level; basically the max valid range
OUTLIER_TOLERANCE = 0.05    # measurements this percentage from other values are considered outliers and discarded

# Define GPIO to use on Pi
GPIO_TRIGGER = 23
GPIO_ECHO = 18

TRIGGER_TIME = 0.00001
MAX_TIME = 0.1  # max time waiting for response in case something is missed
TIME_BETWEEN_MEASUREMENTS = 0.5  # time to sleep between each measurement

GPIO.setmode(GPIO.BCM)
GPIO.setwarnings(False)
GPIO.setup(GPIO_TRIGGER, GPIO.OUT)  # Trigger
GPIO.setup(GPIO_ECHO, GPIO.IN, pull_up_down=GPIO.PUD_UP)  # Echo
GPIO.output(GPIO_TRIGGER, False)

# This function measures a distance
def measure():
    # Pulse the trigger/echo line to initiate a measurement
    GPIO.output(GPIO_TRIGGER, True)
    time.sleep(TRIGGER_TIME)
    GPIO.output(GPIO_TRIGGER, False)

    # ensure start time is set in case of very quick return
    start = time.time()
    timeout = start + MAX_TIME

    # set line to input to check for start of echo response
    while GPIO.input(GPIO_ECHO) == 0 and start <= timeout:
        start = time.time()

    if(start > timeout):
        return -1

    stop = time.time()
    timeout = stop + MAX_TIME
    # Wait for end of echo response
    while GPIO.input(GPIO_ECHO) == 1 and stop <= timeout:
        stop = time.time()

    if(stop <= timeout):
        elapsed = stop-start
        distance = float(elapsed * 34300)/2.0
    else:
        return -1
    return distance
    
# take 3 measurements, discard outliers, return averge result
def sample():
    measurements = []
    attempts = 0

    while len(measurements) < 3 and attempts < 15:
        if attempts > 0:
            time.sleep(TIME_BETWEEN_MEASUREMENTS)
        distance = measure()
        attempts += 1
        if(distance >= DISTANCE_TO_FULL and distance <= DISTANCE_TO_EMPTY):
            measurements.append(distance)

    if len(measurements) == 3:
        # check for and remove outlier measurements
        for i in range(len(measurements)):
            if is_outlier(measurements, i):
                measurements.pop(i)
                
        # if there are at least 2 good data points remaining, take the average and return it
        if len(measurements) >= 2:
            measure_sum = 0
            for i in range(len(measurements)):
                measure_sum += measurements[i]
            measure_avg = measure_sum/len(measurements)
            return (measure_avg)
        else:
            # we didn't have at least 2 good data points
            return -2
    else:
        return -1
        
# run the sample function twice, discard result if both values not within OUTLIER_TOLERANCE of each other
def double_sample():
    trials = [-1, -1]
    
    # run the two trials
    for i in range(len(trials)):
        attempt = 0
        while attempt < 6 and trials[i] < 0:
            attempt += 1
            trials[i] = sample()
    
    # compare the two results
    if is_outlier(trials, 0):
        # the two values aren't consistent
        return -1
    else:
        trial_avg = (trials[0] + trials[1])/2
        return trial_avg

# compare the value of the list item at the given index to every other item in the list
# if the given item varies by more than OUTLIER_TOLERANCE compared to the other items, it's an outlier
def is_outlier(list, index):
    if index > len(list)-1:
        return False
    else:
        outlier_compares = 0
        outlier_true = 0
        value = list[index]
        for i in range(len(list)):
            if i != index:
                outlier_compares += 1
                diff = abs(list[i] - value)
                diff_percent = diff/list[i]
                if diff_percent > OUTLIER_TOLERANCE:
                    outlier_true += 1
                else:
                    break

        if outlier_compares == outlier_true:
            # every other item has a value that is off by at least OUTLIER_TOLERANCE, this is an outlier
            return True
        else:
            # this item is within OUTLIER_TOLERANCE of every other item
            return False
            
# take the measurement
attempts = 0
percent_full = -1
while attempts < 5:
    attempts += 1
    result = double_sample()
    if result >= 0:
        percent_full = (1-(result/DISTANCE_TO_EMPTY)) * 100
        break

GPIO.cleanup()

# Store measurements in database (must specify the channel and measurement)
self.store_measurement(channel=0, measurement=percent_full)
```
4. Save/activate the input

### Expected behavior

From here, I can navigate to Data -> Live and see the latest measurement as expected:
[image]

But if I try to create a graph widget, or view async graph data, they're just blank - presumably because historical data isn't being saved to the database properly:
[image]

Viewing the log, I see that a bunch of StopIteration errors are being thrown on many (but not all, judging by the timestamps) executions:
[image]

### Additional context

This code has been successfully running on a Mycodo 8.8.8 install for a couple weeks without issues. I attempted to use Mycodo's upgrade function to move to 8.9.1, but that broke my install (HTTP 500 errors on most pages, presumably from a dropped custom measurement I had created - I already created a [separate issue](https://github.com/kizniche/Mycodo/issues/949) for that). So I moved my Mycodo installation to a backup directory and installed 8.9.1 from scratch. Just mentioning in case that might play a part in the issue.

kizniche, 2021-03-15:
I have been unable to replicate your issue. These are the steps I used on 8.9.1:

1. Add new measurement with ID "per_full", Name "Percent Full", and Unit "percent"
2. Add Python Code input and set measurement/unit to Percent Full/Percent, leave all other options default.
3. Activate Input.
4. Go to Live page (shows last measurement).
5. Go to Dashboard and add graph widget with input selected and default options (shows past measurements).
6. Go to Dashboard and add measurement widget with input selected and default options (shows last measurement).
7. Go to Asynchronous graph page, select Input, and press Day button (shows past measurements).

kizniche, 2021-03-15:
If the Live page and widgets are displaying data at all, there are measurements being stored in the database. What is strange is your Graph Widget not displaying data. It appears your Input is set up exactly as mine (minus the non-default code). What is missing from your bug report is a screenshot of the Graph Widget options, because this appears to be the only thing displaying an issue.

rbbrdckybk, 2021-03-15:
Thanks for taking a look. I'm very new to Mycodo, so it's possible I'm doing something dumb. The steps you outlined are the same as mine, other than adding my custom Python code.

I just went back and added a new test input and did all the same steps, except I didn't add my custom code (just leaving the default example code). In this case, the graphs work properly. One odd thing I noticed here is that the code evaluation runs for the example code, but it does not when I put my custom code in - in Mycodo 8.8.8, I distinctly remember that the code evaluation did run. In 8.9.1, when I press "save" on the input with my custom code, the page just refreshes without any kind of message (all of my changes do get saved, though).

Adding some additional screenshots for my original input in case they're helpful, although the issue definitely seems to be the inclusion of the custom code:

Here is my graph widget settings - all defaults other than the name:
[image]

And on the async page, if I press any of the time buttons, the page refreshes but no graph or other data appears:
[image]

rbbrdckybk, 2021-03-15:
Turns out I am indeed doing something dumb - everything does work properly. I wasn't selecting the input on either of the graph pages. This only became obvious to me when I added a second input to test with and saw that I needed to click one of the two. With only one input available the UI made it look (to me) that the only input I had was already pre-selected.

Apologies for the confusion!

kizniche, 2021-03-15:
I managed to find an issue with the success response, with was silently failing (an issue with werkzeug). You had so much code that the response was exceeding the cookies limit of 4093 bytes in the response. I changed to server-side sessions that rely on the filesystem, allowing the response with the pylint info to display. I just pushed the fix and it'll be available in the next release unless you want to [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) to test.

kizniche, 2021-03-18:
It looks like there was still an issue with displaying the pylint3 report in 8.9.2 release, but it has been fixed for the yet-to-be released 8.9.3.