kizniche/Mycodo #951: Remote Admin - Can’t add host

FruityLemonade, 2021-03-14:
### Versions:

 - Mycodo Version: [8.9.1]
 - Raspberry Pi Version: [4 B 2/4GB]
 - Raspbian OS Version: [Buster Lite]

### Reproducibility

1. Go to 'Remote Admin'
2. Enter details and click on 'Add Host'
3. After the site refreshes “Error: Error: Remote Host Add: write() argument must be str, not None” is displayed and no host has been added. 

### Expected behavior

A second RasPi running mycodo to be added to the host list for remote administration.

### Screenshots

[image]

kizniche, 2021-03-15:
This is an experimental feature I started to work on and haven't touched it in a while. I thought I had a notice explaining it was experimental and not to submit issues about it, but that may have changed. In any case, this is in its very early stages and I can't guarantee it works. I'm also not very inclined to respond to bug reports because I already know it has a bunch of issues.

kizniche, 2021-04-20:
I've removed the menu entry for Remote Admin so it's not readily accessible in the next release.

kizniche, 2021-04-20:
This issue has been mentioned on **DIY Automation Forum**. There might be relevant details there:

https://forum.kylegabriel.com/t/cant-create-a-remote-admin/110/2

closed 2021-04-24 by kizniche