kizniche/Mycodo #953: Cannot activate Bang-Bang Hysteretic function

JustinCredible-, 2021-03-16:
Trying to Activate a Bang-Bang Hysteretic function after initial configuration and saving. Errors with "Error: Error: receiving: timeout"

 - Mycodo Version: [e.g. 8.0.3] 8.9.1
 - Raspberry Pi Version: [e.g. 3B+] a02082
 - Raspbian OS Version: [e.g. Buster Lite] Linux raspberrypi 5.10.17-v7+

1. Go to Setup -> Function
2. Click on Activate on Bang-Bang Hysteretic function
3. After a period of time, page will refresh with success messages "Success: Function Activate (SQL)
Success: Activate Controller" and error message "Error: Error: receiving: timeout"

Expect function to enable and start working.

[image]

[image]

Brand new installation using SHT31D sensor, all sensor values reporting (temp, humidity, dewpoint, VPD) and PID Controller, Conditional Controllers activate and work fine.

kizniche, 2021-03-16:
Thanks for the bug report. I'll take a look.

JustinCredible-, 2021-03-16:
Sure! I was kind of wondering what logs I should poke at to see where it’s
failing on my side

kizniche, 2021-03-16:
Configure -> Mycodo Logs -> Daemon Log is where I would start. I see the issue and am figuring out a fix.

kizniche, 2021-03-16:
Just pushed a fix that will be available in the next release.

closed 2021-03-16 by kizniche

JustinCredible-, 2021-03-17:
Upgraded to 8.9.2 and the function activates now. Thank you!