kizniche/Mycodo #955: Still Image Path field in camera config

sourcoding, 2021-03-17:
### Describe the problem/bug

I am trying to set the "still image Path" in the webgui to point to a smb share I have mounted on my rasberry PI.
When I insert the path and click on save, it will not save the entry and when I go back to view the field it is empty.

### Versions:

 - Mycodo Version: [8.9.2]
 - Raspberry Pi Version: [3B+]
 - Raspbian OS Version: [Buster]
- Camera Type: [Picamera]

### Reproducibility

1. Go to menu more->cameras
2. Click on 'plus sign next to camera'
3. Insert path in "Still Image Path" field (Ex: /sambashare)
4. Click on save
5. Click on plus next to same camera
6. "Still Image Path" is empty"

### Expected behavior

I was expecting to change the destination directory of where images are stored.

### Screenshots

NA

### Additional context

I have verified that mycodo user can read and write in the new destination directory.

Thanks
Sandro

kizniche, 2021-03-17:
Thanks for the bug report. I just pushed a fix that will be available in the next release.

sourcoding, 2021-03-18:
Happy to help!
Cheers
sandro

closed 2021-04-24 by kizniche