kizniche/Mycodo #958: SHT4x input device

pkrahmer, 2021-03-18:
**Is your feature request related to a problem? Please describe.**
Add Sensirion SHT40 device. E.g. used in this product: https://www.adafruit.com/product/4885

**Describe the solution you'd like**
Using adafruit library, make the new SHT40 device available.
I created an implementation based on the SHT3x circuitpython module. It is running well for my adafruit sensor board.

**Additional context**
Please find implementation in pull request https://github.com/kizniche/Mycodo/pull/957

pkrahmer, 2021-03-19:
It seems my main change on this sneaked in already with #957
I was not aware that any changes I do in my fork after requesting a PR but before it is merged are automatically amended to the existing PR. Hope you also agree to have this device included.
This PR is just one little extension to allow the 4 different i2c addresses the sensor supports via solder jumpers. They are not documented so I tried all combinations with my device manually.

kizniche, 2021-03-19:
As long as it works, I'm fine with it. I'll probably change the metadata that generates the help docs before release, but everything else looks great.

closed 2021-04-24 by kizniche