kizniche/Mycodo #961: Compatibility Question: Inputting data from Atlas Scientific Hydroponics kit

rotorhead412, 2021-03-24:
Can the data from the WIFI Hydroponics sensor kit from Altas Scientific be imported into MyCodo?

Seems the cheapest way to get the necessary sensors with decent reliability.

Awesome program, will be sponsoring once my system is operational.

Cheers 

Bryn

closed 2021-03-24 by rotorhead412

kizniche, 2021-03-24:
The Wifi Hydroponics Kit [setup guide](https://atlas-scientific.com/files/Wi-Fi-Hydroponics-kit-setup-guide.pdf) mentions the code Atlas Scientific provides has the capability of uploading measurements to ThingSpeak. I'm not familiar with ThingSpeak, but they appear to have a robust [API](https://www.mathworks.com/help/thingspeak/read-data-from-channel.html?s_tid=CRUX_lftnav). You can create your own Mycodo Input that queries this data, or yu can use a built-in Input (such as the MQTT Subscribe Input) to acquire the data from ThingSpeak.

alexwine36, 2021-03-28:
Would it be possible to have the board publish the result to the Mycodo api and read in the result as an input?

kizniche, 2021-03-29:
Sure, if you wrote a sketch that sent the data somehow, and uploaded it to the MCU of the Atlas Scientific kit, then interfaced it somehow to the Pi, then developed a Mycodo Input that received and parsed the signal into measurements. This would be far more reliable than depending on wifi, a router, and an internet connection, but at this point, it wouldn't be much different from just using the sensors alone and connecting them all to the Pi. The only difference is the first method requires you to develop both an MCU sketch and a Mycodo Input, whereas the second method only requires you to connect all the sensors to the Pi because there already exists Atlas Scientific Inputs to read directly fro the sensors.

kizniche, 2021-03-29:
I think the I2C pins are broken out on the board of the kit, so you could connect the Pi directly to the board. You could also write a sketch that publishes the measurements over wifi to an MQTT server running on the Pi and use the Mycodo MQTT Subscribe Input to get the measurements.

rotorhead412, 2021-03-29:
Abandoned the idea due to lack of peristaltic pump outputs. Bit the bullet and got it all individually. Very keen to rig up the sensors and start the dive into PID control.