kizniche/Mycodo #962: Support for I2C or SPI controlled potentiometers for physical control of humidifier output?

LucidEye, 2021-03-25:
Not sure if this is the correct place to ask this question, so I apologize in advance if I have broken protocol :-P

Does Mycodo have built-in support for I2C or SPI controlled digital potentiometers?
I do not see such devices listed in the "Supported Outputs" section of the manual.

I would like to be able to use I2C or SPI controlled potentiometers to automate the control of devices.
Specifically in this instance I am using a cheap ultrasonic humidifier to maintain RH and VPD in a hydroponics system.
I live in a very hot, dry climate and the humidifier is an absolute must-have component for maintaining ideal environment.  However, the only way to control it's output is via a manual knob (analog potentiometer).  
I would love to replace that analog potentiometer with a remotely-controllable digital potentiometer such as this...

 https://learn.adafruit.com/ds3502-i2c-potentiometer

Right now the only way I can control my RH is to increase or decrease the intake & exhaust fan speeds in my hydroponics system (via PWM).  While this does work to some extent, the RH takes too long to balance and reach desired levels, and I have to constantly adjust the output on the humidifier manually several times a day to get the RH just right.  

It would be a HUGE advantage to be able to use a Mycodo widget to physically control my humidifier output via automation in conjunction with a temp/humidity/pressure sensor rather than having to manually fuss with analog knobs all day. :-(

There are several devices that could benefit from the use of digital potentiometers:
Fan speed controllers (as an alternative to PWM).
Light dimmers.
Humidifiers.
Water chillers.
Tank stirrers. 
DC water pumps (to control flow rates)
Etc. etc.

Is this something that Mycodo already does, or is it something that I would have to custom program?  Unfortunately I am not a programmer, and I am really looking for control software that is mostly "plug and play" with some relatively simple calibration rather than spending weeks to figure out how to write and integrate the code for such a device. :-(

By the way, Mycodo is an AMAZING project, and I can't wait to take my hydroponics system design to the next level using it's automation!  THANK YOU TO EVERYONE for creating this incredible software! <3

Thank you,
JCL

kizniche, 2021-03-25:
There currently aren't any built-in digital potentiometer Inputs, but I do think it is an interesting device. If you had a device to test, I could provide a custom Input module you can import and test.

Alternatively, if you haven't already considered it, you can modulate the power to your humidifier to achieve a similar result. I will typically connect my humidifier to a solid state relay and use that with a humidity seneor in a PID controller, which will turn the humidifier on for a duration of time every PID period.

Magnum-Pl, 2021-03-25:
As an alternative to the digital potentiometer you can use the MCP4728 DAC to get four separate 0-5V outputs. I've been using the MCP4728 to control a couple analog potentiometers on a water chiller and it works great. 

It's also possible to use an MQTT analog output with something like an ESP8266 to get a wireless 0-5V output, but I haven't tried that yet.

LucidEye, 2021-03-25:
Thanks for all these suggestions, but they don't seem like really viable "alternatives" to simply replacing an analog potentiometer with a digital one of the same resistance range.
My goal is to be able to set a target range for the RH and have the software automatically "learn" and adjust the fans and the humidifier output to maintain the most level RH & VPD without a lot of up & down spikes.  
The secondary goal is to maximize efficiency and reduce the water use in the humidifier so I don't have to fill it so often. 
The third goal is to reduce the amount of ingress into the grow space as much as possible to reduce the introduction of possible contaminants (dust, dirt, mold spores, insects, etc.).  I'm going to be installing hepa air filters on the intake fans when I do my automation upgrade-rebuild ;-)
My ultimate goal is to take what I learn from automating my current tiny grow space, and scaling it up to work in the greenhouse I plan on building this summer :-)

@ kizniche
"Modulating" the humidifier with just a simple on/off relay isn't going to give me the precise, rapid control I need, I know this because that's what I am already doing ;-).  And depending on the temp/RH of the incoming air it either takes too long to reach target levels, or it causes wide up & down swings in the RH instead of maintaining a nice steady level.... hence the need to constantly fiddle with the knob on the humidifier  ;-)  I'm also worried that constantly cycling the power on and off may shorten the life of the humidifier as opposed to ramping it's output with the potentiometer :-(
My thoughts were to use 2 temp/humidity sensors, one outside the grow space, and one inside to try to more precisely calculate the humidifier output setting to balance RH in the grow space.

If you would be willing to provide a custom input/output module for that Adafruit unit that I could test that would be awesome... I'm better at tinkering with already written code than I am at writing it from scratch :-P.  And obviously it would be easier to use a potentiometer that already has libraries written for the Pi ;-)  I was actually kind of surprised that this hasn't already been implemented in Mycodo being that it is one of the most common ways to adjust input/output on may electronic devices.  I think if we could work together to make it happen, a LOT of people would find it very useful.... I'm certainly willing to put some time into it if it would add a whole new realm of automation control possibilities into Mycodo :-)

@ Magnum-Pi
I actually considered using a DAC, but as you mentioned, that only gives me 4 increments of control, I really want the infinite variable control of a potentiometer for more precise ramping of the humidifier output.  Also, using a DAC seems to require a lot of extra custom circuit design to implement it :-(.  I need a simple drop-in replacement, and simply replacing the existing analog potentiometer with a digital one of the same ohm rating seems like a no-brainer :-P  I like to design things using the KISS principal (Keep It Stupid Simple) ;-)

Thanks for such quick replies :-)

kizniche, 2021-03-26:
>  I'm also worried that constantly cycling the power on and off may shorten the life of the humidifier

I've ran multiple ultrasonic transducers off solid state relays continually for years without them failing. I've actually had the fan (6 years) and power supplies (underpowered for a 10-transdicer unit) die, but not the transducers.

Here's a quick DS3502 Output module that can be imported into Mycodo to test with. Rename the extension to .py and import on the Configure -> Custom Outputs page.

[module_ds3502_01.txt](https://github.com/kizniche/Mycodo/files/6208558/module_ds3502_01.txt)

rotorhead412, 2021-03-26:
I'm experiencing a similar problem, attempting to link up a ceiling fan speed controller intended for the wall of a house to Mycodo. Keeping in line with KISS, could you not just attach a stepper motor to the  analogue potentiometer? I am shooting for simply 3 positions on the steppers plain of rotation, for a low, medium and high setting. In your case you could program the motor to move a desired angle (in steps) forward or backward  depending on an input.  

Just thought i'd put it out there :) Figuring out how to do that is my current task.

kizniche, 2021-03-26:
> attach a stepper motor

I recently added stepper motor support to Mycodo. It's a generic output module that should work with virtually all standard stepper motors.

LucidEye, 2021-03-26:
Wow, that was fast! Thank you so much kizniche!!!  I'll be digging into this a soon as I'm finished digging in our outdoor garden... gotta get all our veggies planted this weekend :-P

Magnum-Pl, 2021-03-26:
>I actually considered using a DAC, but as you mentioned, that only gives me 4 increments of control

I said the MCP4728 has four separate outputs,  not four increments. Each output channel has 1023 increments between 0-5V.

>using a DAC seems to require a lot of extra custom circuit design

The MCP4728 is available in a ready to use module for less than $10 on amazon. 

Connecting a stepper to the potentiometer sounds interesting. Ive seen some motorized potentiometers before, but they were pricey.

LucidEye, 2021-03-26:
Magnum-Pi

My apologies for misunderstanding :-P.  I have to dismantle my humidifier again and do some more probing and testing on the potentiometer control circuit to see what the voltage is actually going through it.  That was another reason I decided against the DAC because of it's lower operating voltage (and the need to build other circuitry around it to make it conform to the operating voltages in the humidifier).  It's easier to find a d-pot that is rated for wider voltage ranges and different ohm ratings that can just be "dropped-in".  I could even add a switch to flip back and forth from manual control to digital control... so I can use the humidifier as it was originally configured, or flip the switch and hand it over to the automation system.  I like having analog "manual override" options, especially in the hydroponics system for things like turning on the pump when it's time to drain and change the nutes in the tank ;-)

The stepper motor idea is also a feasible option, I have a little experience building motion-control units with 28BYJ-48 motors for time-lapse cameras using the R-Pi.  But the physical implementation can be a pain and seemed like way too much work just to turn a knob... modifying the housing of the humidifier, fabricating a bracket to hold the motor in the right place (I don't own a 3D printer... yet), having to use a motor-controller hat on the Pi and having it block the GPIO pins, etc.. There was also the problem of trying to figure out "where is that knob actually set" using an arbitrary "number of steps" or motor shaft angle of rotation, especially if there is any backlash in the motor when changing directions like in the 28BYJ-48 :-P .  It's so much easier and accurate to just use the resistance value of the d-pot wiper as a direct correlation to the output of the humidifier ;-)

LucidEye, 2021-03-27:
kizniche

What solid-state relay boards are you using to run your humidifiers?  And what is their max amperage on the AC side?  

I'm looking for solid state relay boards to run my:
LED grow lights.
Watering pump.
Ventilation fans.
Humidifier.

I know the common 2A solid state relays can easily handle my pump, fans, and humidifier, but I doubt they can handle my LED lights.
Any suggestions or sources for something closer to a 4 or 6 amp solid state relay would be much appreciated.

Thank you again.
JCL

kizniche, 2021-04-03:
> What solid-state relay boards are you using to run your humidifiers? And what is their max amperage on the AC side?

My humidifiers have a very low power draw, and they're resistive loads, so I've used these without issue:

 - 4-channel: https://www.amazon.com/dp/B00ZZVQR5Q
 - 8-channel: https://www.amazon.com/dp/B00ZZW7MI6

There are also off-brands with the same components that are a bit cheaper.

For larger loads, I usually go to digikey and start filtering solid state relays for my specifications. In your case, select 3.3 voltage input, 4 to 6 amp load, 110 (or 220) VAC voltage load, then sort by price. Here's what I got: https://www.digikey.com/short/3hbr9495

You can then consider form factor and mounting type. I have bought a lot of Crydom products and like their chassis mount SSRs. On that list, here are the least expensive:

- 5 Amp ($31): [CL240D05](https://www.digikey.com/short/rt2j312f)
- 10 Amp ($33): [CL240D10](https://www.digikey.com/short/995q7127)

closed 2021-09-20 by kizniche