kizniche/Mycodo #965: Mycodo Discord

kizniche, 2021-03-26:
Although there still exists a private discord room only for supporters, I've opened up the mycodo discord room to the public. You can find a link to join in the readme at https://github.com/kizniche/Mycodo#discussion

MangoMars23, 2021-05-08:
Hello. Wow, thanks for Mycodo! Please with the BT sensor Xiaomi: Mijia LYWSD03MMC i have problem to see measurement data. I see inlu 0 (zeros) in Live Data. I have found MAC address trough the original Xiaomi App and pasted it to the Mycodo. Is there any other setting to make the sensor work in Mycodo? Thank you and Big Up!
[image]
[image]

kizniche, 2021-05-08:
Please keep issues separate by creating a new Issue.

FruityLemonade, 2021-05-08:
Hey, there’s two possible issues I see: 
1) You have ATC-Mode enabled In the input. Have you flashed your sensor with the ATC firmware? If not, I highly recommend doing it for better functionality than with the stock firmware: https://github.com/atc1441/ATC_MiThermometer
With the stock firmware a connection can only be established by mycodo if the device has never been initalized through the Xiaomi app, otherwise the sensor encrypts it’s BLE advertisements and the non-ATC input module doesn’t have the option to add this key to decrypt the advertisements. 

2) are you sure hci1 is correct? The default built-in Bluetooth adapter is hci0 on my raspberry 4.